ARM-software/cmsis-pack-eclipse #2: UI components no longer reusable

jesperes, 2015-10-16:
The 0.9 version of the plugin had a set of "reusable" UI components which was very useful and made the plugins usable even in an environment where the Eclipse workbench has not been created.

In 1.0, there is no such distinction any more. The .widgets package *seems* to be intended to be usable even outside the Eclipse workbench, but makes references to workbench shared images.

What is the policy regarding requiring a running workbench? I can of course get around this problem for the time being, but I would like to avoid complications in coming releases.

edriouk, 2015-10-20:
Hello Jesper,
Our intention was to provide images consistent with running platform. Therefore we tried to use shared images where possible. Besides we have not considered possibility to run the plug-ins without workbench. Apart from shared images, the .ui plug-in needs workbench to access command and browser services.
We could consider replacing the shared images with local ones in next releases.
Which widgets in particular are you interested in? 
What other improvements would you suggest to make the plug-ins more usable?
Besr regards, 
Evgueni

jesperes, 2015-10-20:
I'm primarily interested in component selection/validation and device selection. I'm trying to integrate this with IAR Embedded Workbench (which is a plain C++/MFC application), but I would like to avoid requiring the user to have a running Eclipse workbench. So my current plan is to run a headless Eclipse application in the background which will open plain SWT shells. Currently I have a functioning RTE editor based on a SWT shell which hosts a `RteComponentManagerWidget`. The only problems that I've discovered so far is the use of PlatformUI shared images and browser support. It's ok for my part to have dependencies to Eclipse UI plugins, but not to require that the workbench is started.

The `RteDeviceSelectorWidget` and possibly the `RtePackSelectorWidget` would also be of interest. Also, pack downloading is a feature that we will need in some form. Not sure yet how we will approach that.

/Jesper

edriouk, 2016-01-20:
Hello Jesper,
The release 1.1.0 isolated selection/validation and device selection dependencies on workbench.
Icons are all local to the plug-in.  
You will need to implement  IRteModelController interface (extend RteModelController implementing openUrl() method) to connect the component selector widget with your environment.

edriouk, 2016-03-18:
I believe the problem is fixed in release 1.1, closing issue

closed 2016-03-18 by edriouk

edriouk, 2016-07-27:
Hello Jesper, could you please test if the release 2.0 works for your environment. 
We have put attention on wrapping PlatformUI calls, but we might have overlooked some places.

jesperes, 2016-07-27:
Ok, I will do that.