ARM-software/cmsis-pack-eclipse #4: Creating an empty RTE configuration

jesperes, 2015-11-02:
I'm trying to figure out how to set up things so that I can open simple SWT shell containing the RteDeviceSelectorWidget to select a device, without having anything to start from. This is as far as I can get:

        IRteModelController controller = new RteModelController(new RteModel());
        ICpConfigurationInfo config = new CpConfigurationInfo();
        ICpDeviceInfo deviceInfo = new CpDeviceInfo(config);
        config.addChild(deviceInfo);
        controller.setConfigurationInfo(config);

But when calling `setConfigurationInfo` I get stuck on 

    java.lang.NullPointerException
 	at com.arm.cmsis.pack.info.CpDeviceInfo.updateInfo(CpDeviceInfo.java:141) 
 	at com.arm.cmsis.pack.info.CpDeviceInfo.setRteDevice(CpDeviceInfo.java:92)
	at com.arm.cmsis.pack.rte.RteModel.resolveDevice(RteModel.java:195)
	at com.arm.cmsis.pack.rte.RteModel.update(RteModel.java:131)
	at com.arm.cmsis.pack.rte.RteModel.setConfigurationInfo(RteModel.java:121)
	at com.arm.cmsis.pack.rte.RteModelController.setConfigurationInfo(RteModelController.java:171)

Also, I cannot seem to figure out how to do this using the wizards/editors in Eclipse either. There does not seem to be a wizard, and creating an empty file with the `.rteconfig` extension just cause XML parsing exceptions.

edriouk, 2015-11-04:
Thank you for the reporting the NPE. It will be fixed in the next release.

RTE project creation is done by CreateRteProject template process. 
It is  triggered by executing New Project wizard: File->New C Project.
Select CMSIS RTE C/C++ Project template.
You have to implement a toolchain adapter for you compiler. But to experiment with the project management you can use the generic one. It will also help you to develop your own adapter.

jesperes, 2015-11-04:
The problem in my case is that I'm trying to use the RTE functionality in a
context where I don't have an Eclipse project. I'm investigating whether I
can reuse these plugins to implement CMSIS-functionality in our Windows
IDE, and in that case I just want to create a RTE configuration for use
with our own project file format.

We will in time also implement an adapter for our own Eclipse-based
toolchain(s), but we're still in the prototyping phase.

BTW, I was sorry to miss the talk at EclipseCon yesterday.

/Jesper

edriouk, 2015-11-04:
In fact, you do not need to use the .project plug-in. 
Most of GUI such a widget to select a device is generic.The RTE configuration editor has no dependency on Eclipse project at all. 
You might look how RteConfiguration class works and implement similar functionality in our own plug-in. In that case you would not even need a toolchain adapter.

Are you attending the EclipseCon? If so, we can meet  and I can give you detailed look in the plug-in.

jesperes, 2015-11-04:
Ok, thanks. I will try that.

I will not be at EclipseCon Europe this year, but I was planning to go to
EclipseCon NA next year. Maybe we can meet there?

/Jesper

edriouk, 2016-01-20:
Hello Jesper,
FYI : I'll be available at Embedded World Exhibition (23.-25.2.2016 Nuerenberg, Germany) all three days.
We could meet there for discussions.

jesperes, 2016-01-20:
I probably won't be there, but I'll check with my colleagues if anyone is going.

marek-trmac, 2016-03-17:
It would be nice to have such capability in Eclipse UI too. It would be useful for existing project to add RTE nature.

edriouk, 2016-03-17:
Hi Marek,
Adding an RTE nature to an existing project might be implemented in future.
The first aspect here to consider is:
An existing project is probably already set for a specific device. That device should be respected when adding the RTE nature.  But how to get this information in a generic way?

marek-trmac, 2016-03-17:
My proposal would to do it same as new project wizard - user should select device.

edriouk, 2016-03-18:
The initial issue was about handling of RTE configuration. I believe it is solved, closing issue.

closed 2016-03-18 by edriouk