ARM-software/cmsis-pack-eclipse #5: 'RTE Project Updater' has encountered a problem

ilg-ul, 2015-11-16:
This problem was also present in version 0.9, I already reported it, but apparently it got lost:

> Error updating RTE project
Cannot create linked resource.  Local location '/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipses/eclipse-44-cmsis2/Eclipse.app/Contents/MacOS/file:/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages/Keil/STM32F4xx_DFP/2.6.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c' does not exist.

The `.../file:/...` present in the middle of the path suggests a string processing problem.

This error prevents the creation of projects.

edriouk, 2015-11-17:
Hi Liviu,
Thank you for the feedback. I cannot reproduce this issue.  Could you send me a CMSIS Pack root path that you use. I suspect that is in the same location as your workspace.  I have not tested such situation.

ilg-ul, 2015-11-17:
`/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages`

as you guessed, it is at the same level as the workspace.

the RTE/Device/STM32F407VGTx/startup_stm32f407xx.s, stm32f4xx_hal_conf.h and system_stm32f4xx.c were copied to the project, the RTE_Components.h was created, but for all other sources the plug-in failed to refer/copy.

edriouk, 2015-11-17:
I'll have a look in this issues. Meanwhile could you please use CMSIS pack root directory outside the workspace.

ilg-ul, 2015-11-17:
it is outside the workspace, `Packages` and `workspace` have the same parent.

edriouk, 2015-11-18:
Hi Liviu,
do you use symbolic links in your setup? In particular, does ${cmsis_pack_root} point to a physical location or to a symbolic link?

ilg-ul, 2015-11-18:
nope.

ilg-ul, 2015-11-18:
please take a closer look to the (very long) path in the first message. the part after file: is correct, but you concatenate it with the first part, and this makes it an incorrect path.

edriouk, 2015-11-18:
The prefix before file:/ corresponds to Eclipse executable location. I have no idea how it gets prepended. 
All paths supplied to create a link start with ${cmsis_pack_root}. Nothing gets prepended to it before that point. This variable gets expanded during link creation process (package org.eclipse.core.internal.resources).  
We see that the variable gets expanded correctly and only then the Eclipse executable path is added.
The only explanation could suggest is that the expanded sting is treated as relative to current working directory (executable path). 
I still cannot reproduce the problem, neither on Windows nor on Linux. 
Could it be a Mac specific issue or your specific setup?
Could you try to create a linked resource manually?

ilg-ul, 2015-11-18:
> I still cannot reproduce the problem, neither on Windows nor on Linux. 

could you also try on a Mac?

if you have any idea how I can help, please let me know; I packed the entire project and published it on [Dropbox](https://dl.dropboxusercontent.com/u/78151643/t2.zip)

> Could it be a Mac specific issue?

hard to say how probable it is, but yes, it is possible

> or your specific setup?

highly unlikely

> Could you try to create a linked resource manually?

I never had problems to create linked resources from the Eclipse interface; if you suggest a source and a destination, I can try again.

edriouk, 2015-11-18:
I have tried to work with your project on Windows and have even managed to build it (though I have had to resolve some toolchain issues).  But all linked resources are inserted/updated correctly.
You could probably help to locate the problem by debugging it:
1) install CMSIS-Pack plug-in including source code in your Eclipse development  environment (you do not need to clone source repository, just install the pre-built feature). 
2) Set a breakpoint in RteProjectUpdater.AddFile() method where it calls ProjectUtils.createLink(project, srcFile, dstFile, monitor);  - use Search->Java to locate the method
3) Start debug session and create a CMSIS RTE project, select a component that produces links  
4) On breakpoint hit check if srcFile already get corrupted
5) if not, track down link creation ... 
Thank you for your feedback and help

ilg-ul, 2015-11-18:
> Search->Java to locate the method

this search does not find your method. 

I have a separate Eclipse with CDT, Java, Eclipse development, etc, and I can start debug sessions, but I could not find a way to access the sources and to set the breakpoint.

ilg-ul, 2015-11-18:
I managed to run a debug session (using the git sources), and the exception is raised in `Resource.createLink()`

the `URI localLocation = $%7Bcmsis_pack_root%7D/Keil/STM32F4xx_DFP/2.6.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma.c`

the macro is resolved in `ProjectPathVariableManage.resolveURI()`, to `file:/Users/ilg/My%20Files/MacBookPro%20Projects/GNU%20ARM%20Eclipse/Eclipses/eclipse-44-rcp-cdt/Eclipse.app/Contents/MacOS/file:/Users/ilg/My%20Files/MacBookPro%20Projects/GNU%20ARM%20Eclipse/Eclipse%20Workspaces/Packages`

which is obviously wrong.

from here to the exception it is just a matter of time:

```
!MESSAGE Error updating RTE project
!STACK 1
org.eclipse.core.internal.resources.ResourceException: Cannot create linked resource.  Local location '/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipses/eclipse-44-rcp-cdt/Eclipse.app/Contents/MacOS/file:/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages/Keil/STM32F4xx_DFP/2.6.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma.c' does not exist.
	at org.eclipse.core.internal.resources.Resource.assertLinkRequirements(Resource.java:195)
	at org.eclipse.core.internal.resources.Resource.createLink(Resource.java:673)
	at org.eclipse.core.internal.resources.Resource.createLink(Resource.java:647)
	at com.arm.cmsis.pack.project.utils.ProjectUtils.createLink(ProjectUtils.java:242)
	at com.arm.cmsis.pack.project.RteProjectUpdater.addFile(RteProjectUpdater.java:284)
	at com.arm.cmsis.pack.project.RteProjectUpdater.addResources(RteProjectUpdater.java:243)
	at com.arm.cmsis.pack.project.RteProjectUpdater.addResources(RteProjectUpdater.java:230)
	at com.arm.cmsis.pack.project.RteProjectUpdater.runInWorkspace(RteProjectUpdater.java:111)
	at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:38)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)

```

edriouk, 2015-11-19:
Thank you Liviu,
The stack shows that the CMSIS-plug-in provides correct input to Resource.createLink()
The variable gets expanded in CpVariableResolver.getValue(), have you checked it works correctly?  
If so, then problem is somewhere else, outside the CMSIS-Pack plug-in...

ilg-ul, 2015-11-19:
`CpVariableResolver.getValue()` returns `file://Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages`

further tracing I reached `UnixFileSystem.resolve()` where the URI lost one of the slashes: `file:/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages`

after some magic, the resolved value is `file:/Users/ilg/My%20Files/MacBookPro%20Projects/GNU%20ARM%20Eclipse/Eclipses/eclipse-44-rcp-cdt/Eclipse.app/Contents/MacOS/file:/Users/ilg/My%20Files/MacBookPro%20Projects/GNU%20ARM%20Eclipse/Eclipse%20Workspaces/Packages`

which is wrong.

the stack trace is:

```
UnixFileSystem.resolve(String, String) line: 108	
UnixFileSystem.resolve(File) line: 133	
File.getAbsolutePath() line: 556 [local variables unavailable]	
URIUtil.toURI(IPath) line: 92	
ProjectPathVariableManager.resolveVariable(String) line: 212	
ProjectPathVariableManager.resolveURI(URI) line: 274	
LocationValidator.validateLinkLocationURI(IResource, URI) line: 106	
Workspace.validateLinkLocationURI(IResource, URI) line: 2562	
File(Resource).assertLinkRequirements(URI, int) line: 177	
File(Resource).createLink(URI, int, IProgressMonitor) line: 673	
File(Resource).createLink(IPath, int, IProgressMonitor) line: 647	
ProjectUtils.createLink(IProject, String, String, IProgressMonitor) line: 242	
RteProjectUpdater.addFile(IRteConfiguration, String, ICpFileInfo) line: 284	
```

Evgueni, I'm afraid you have to find a Mac and run the debug sessions yourself, it is quite complicated to help you further.

ilg-ul, 2015-11-19:
I just checked the same bug on Ubuntu 14 LTS x64, with Eclipse Mars 1 and Java 1.7.0_85, and the result is:

```
Cannot create linked resource.  Local location '/home/ilg/Desktop/GNU ARM Eclipse/file:/home/ilg/Desktop/GNU ARM Eclipse/Packages/Keil/STM32F4xx_DFP/2.6.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dcmi_ex.c' does not exist.
```

I guess the problem affects all non-Windows platforms.

edriouk, 2015-11-23:
Eclipse Mars 1 on native Linux Mint 17.2 Rafaela: Cannot reproduce the problem. 
Could you please do me a favor, try to create the following links manually (File->New->Other->General->File and paste the string into edit under "Link to file in the filesystem" check):
1) ${ECLIPSE_HOME}/notice.html
2) ${cmsis_pack_root}/Keil/STM32F4xx_DFP/2.6.0/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma.c
Check if both links are correctly created.

edriouk, 2015-11-26:
I have tested the plug-in on Mac (10.11.1) with Eclipse Marc1 for C/C++.
The plug-in works as expected, all links are successfully created. 
There should be another cause for the reported problem.

ilg-ul, 2015-11-26:
sorry, I cannot confirm, I removed the plug-ins from my development environment, since they were throwing exceptions each time I started a debug session.

edriouk, 2016-01-20:
Please try release 1.1.0. I have made some enhancement in variable resolver, it might resolve the problem

ReinhardKeil, 2016-03-17:
@ilg-ul Is this problem finally solved or can you still reproduce it?

ilg-ul, 2016-03-17:
the problem is still here:

[image]

edriouk, 2016-07-27:
Hi Liviu,
Could you please have a look if this issue is still exists in v2.0

ilg-ul, 2016-07-27:
will do, but it might take some time.

edriouk, 2016-12-02:
Hi Liviu,
We have tested the plug-in version 2.0.1 at Mac (OS X 10.11.6) with Eclipse Neon and could not reproduce the problem. Could you please check if it still exists for you.

closed 2019-02-22 by ReinhardKeil