ARM-software/cmsis-pack-eclipse #7: Missing linker script?

ilg-ul, 2015-11-16:
I don't know if the problem is related to the 'RTE Project Updater' problem (apparently not), but my test project does not include a linker script (nor an empty main(); nor a missing _exit() function referred from the library).

The result is that the project is not ready to build, as it should usually be.

I'm not sure this is a problem with the CMSIS PACK plug-ins implementation, I think it is rather a CMSIS PACK design problem, since it does not yet provide full support for GCC projects.

Regards,

Liviu

edriouk, 2015-11-17:
Hi Liviu,
This is not a plugin problem. CMSIS-Pack support  for GCC is weak now. 
Linker scripts are only included in startup components for native ARM cores.
Besides, most of the middleware works only for ARMCC.
Regards,
Evgueni

ilg-ul, 2015-11-17:
> CMSIS-Pack support for GCC is weak ... most of the middleware works only for ARMCC

Then, at least for GNU ARM Eclipse, and for now, these plug-ins (or CMSIS PACK in general), remain of little use. :-(

edriouk, 2016-03-18:
I'm closing this issue here as it is not a plug-in problem, but packs.

closed 2016-03-18 by edriouk