ARM-software/cmsis-pack-eclipse #10: The RTE Components tab does not show properly on Ubuntu 14 LTS

ilg-ul, 2015-11-19:
The check buttons in the `Sel.` column do not show at all, the coloured background is painted, but the buttons are not. Clicking blindly triggers some actions, but it is hard to work in these conditions.

Also the horizontal lines are not marked, but this does not impair functionality.

I guess you used some fancy options to paint the widgets, options not available on Linux; perhaps you should be more conservative on some platforms.

edriouk, 2015-11-19:
Could you please send me a screenshot?

ilg-ul, 2015-11-19:
[image]

edriouk, 2015-11-20:
Ubuntu is known to have problems with Eclipse (see for instance 
http://blog.vogella.com/2014/02/21/using-eclipse-luna-on-ubunbu/)
Thus this is not plug-in issue, but GTK

ilg-ul, 2015-11-20:
> Thus this is not plug-in issue, but GTK

you're probably right, but the point is that sometimes, for portability reasons, you should avoid fancy features. 

what do you think would do an Ubuntu user faced with this problem? give up using his beloved Ubuntu? or give up using your plug-ins?

I tried to start Eclipse with 

```
export SWT_GTK3=0
export UBUNTU_MENUPROXY=0 
```

and this solved the problem of grey horizontal lines, but the buttons still do not show up.

edriouk, 2015-11-20:
The problem is probably the same as reported here: 
https://bugs.eclipse.org/bugs/show_bug.cgi?id=435773 (which itself is a duplicate)
Since we use only official APIs, and  nothing "fancy", we would expect the APIs to work correctly. 
Providing a workaround for each particular broken platform does not seem to be reasonable.

ilg-ul, 2015-11-20:
> Since we use only official APIs, and nothing "fancy", we would expect the APIs to work correctly. 

welcome to the beautiful world of Eclipse :-)

> Providing a workaround for each particular broken platform does not seem to be reasonable.

then just be sure you clearly document that your plug-ins are not portable, and list the supported platforms (probably Windows only).

edriouk, 2015-11-20:
In fact, the most valuable part, com.arm.cmsis.pack plug-in, is almost Eclipse-free. That will work on any  platform running Java with only minor changes. 
As far as com.arm.cmsis.pack.project  and com.arm.cmsis.pack.ui plug-ins are concerned, they play more or less "example" role showing how to use the core functionality.

edriouk, 2015-11-23:
I have tried Eclipse Mars 1 on native Linux Mint 17.2 Rafaela:
export SWT_GTK=0 helps - the checkboxes and other decorations are seen in the components tab.

The horizontal lines are not  shown, but they are not shown in any table (e.g. Preferences->C/C++->File Types). Alternating background is used instead, also in the component table.

closed 2016-03-18 by edriouk