ARM-software/cmsis-pack-eclipse #11: Accessing <api/> element in RTE configuration file

jesperes, 2015-11-20:
I'm wondering what the right way is to access the <api/> elements in the RTE config is. There does not seem to be a direct api call to do it, and I'm unsure where to start.

edriouk, 2015-11-20:
RTE config file is parsed into ICpConfigurationInfo object. 
It extend ICpItem => all access methods are available there. In fact in most cases it is enough to work with ICpItem interface.
In particular, accessing  <api> items (null checks are omitted in the snipped for simplicity):
ICpConfigurationInfo config = ...
ICpItem apis = config.getApisItem();
for(ICpItem child :  apis.getChildren()) {
        ICpComponentInfo apiInfo = (ICpComponentInfo)child;
        ICpComponent api = apiInfo.getComponent(); //  api is just a type of a component 
        if(api != null) {// otherwise it is not resolved (not yet, or required pack is missing)
         boolean bExclusive=  api.attributes().getAttributeAsBoolean("exclusive", false); 
          // process files
          ICpItem files = api.getFirstItem("files");
          for(ICpItem f : files.getChildren()) {
              String absPath = f.getAbsolutePath(f.getName());
              ...
          }
       }
Please also have a look in RteModel and RteConfiguration classes which process ICpConfigurationInfo accessing components and apis.

jesperes, 2015-11-26:
Ok, thanks.

closed 2015-11-26 by jesperes