ARM-software/cmsis-pack-eclipse #13: Should CpItem implement equals() and hashCode()?

mario-pi, 2016-01-11:
Since each item is supposed to be uniquely identified by its id, should the `CpItem` default implementation override `Object.equals()` and `Object.hashCode()` based on the result obtained by `ICpItem.getId()`?

This would make it easier to manage a `Collection` of packs e.g. avoiding duplicates.

edriouk, 2016-01-11:
Currently only CpExpression class overloads `Object.equals()` and `Object.hashCode()`.
It is not desirable to be implemented such functionality in CpItem: 
- many ICpItem-derived classes do not  need an ID. 
- the default implementation of ID simply uses getTag() => several distinct items can have the same ID

As far as CpPack is concerned, we could consider overloading  `Object.equals()`,  but  we should consider several spects:
- packs with the same ID can have different PackState ( reserved for future pack installer functionality). 
Such ICpPack objects are not equal, as they can provide different content (some sections are ignored) and reside in different directories.
- CpPack and CpPackInfo  classes use the same ID, but represent different CpItem-based objects.
- by default we are only interested in the latest installed pack => we need to search for pack family first and then for the latest version. CpPackCollection and CpPackFamily classes address that task and provide suitable pack sorting using TreeMap collections.

mario-pi, 2016-01-12:
Thanks for the thorough explanation. It is probably better to leave things as they are now, and implement equality outside of CpItem / CpPack according to the specific use case.

Closing the issue.

closed 2016-01-12 by mario-pi