ARM-software/cmsis-pack-eclipse #14: NullPointerException in CpDeviceitem

bjorniuppsala, 2016-01-13:
CpDeviceItem.getEffectiveChildren() generates a NPE when being run on the family tag in the following .pdsc snippet:
```xml
  <devices>
    <family Dfamily="Family" Dvendor="Vendor">
      <device Dname="DeviceName">
        <processor Dcore="Cortex-M0" Dendian="Configurable" Dfpu="NO_FPU" DcoreVersion="r3p1"/>
      </device>
    </family>
</devices>
```
if <family> contains something like a <description> or any other child the NPE does occur.

The reason seems to be that hasChildren() returns true (since it is using getEffectiveChildCount() ) although getChildren() returns null.

possible patch: https://gist.github.com/bjorniuppsala/0a93d88ff64e80e0dce1

edriouk, 2016-01-20:
fixed in release 1.1.0

bjorniuppsala, 2016-01-21:
Brilliant Thanks!

closed 2016-03-04 by bjorniuppsala