ARM-software/cmsis-pack-eclipse #16: No handling of aliases for Dvendor in <require> tags

bjorniuppsala, 2016-03-04:
I tried to use a pack with the following device (with names changed to protect the innocent)
```XML
    <devices>
        <family Dfamily="MyFamily" Dvendor="Long Vendor:100">
           <device Dname="MyDevice">
    </devices>
```

and a condition:
```XML
    <condition id="SomeCondition">
      <require Dvendor="LV:100" />
    </condition>
```
and a component:
```XML
    <component Cclass="Device" Cgroup="Startup" Cversion="1.0.0" condition="SomeCondition">
      ... 
    </component>
```
But in an RTE with MyDevice choosen, the Deivce/Startup component does not show up.

It seems to boil down to CpConditionContext not using com.arm.cmsis.pack.DeviceVendor to match vendors. I made CpConditionContext extend CpAttributes, which seems to do exactly that.

/Björn

edriouk, 2016-03-04:
Than you Björn for spotting this. The change you've proposed will be included in the next  release.

bjorniuppsala, 2016-03-04:
Brilliant! 

I heard rumors about you visiting the IAR booth at Embedded World to have a look at what we've done...

closed 2019-02-22 by ReinhardKeil