ARM-software/cmsis-pack-eclipse #17: Remove RTE nature from the project

marek-trmac, 2016-03-17:
It would be nice to be able to remove RTE nature from the Eclipse project.
I have removed {project}.rteconfig file using Delete command from popup menu (in Eclipse Project Explorer) and now if I open the project, the following exception is thrown:

```
org.eclipse.core.runtime.CoreException: Error loading RTE configuration file 'rte.rteconfig' parsing failed 
	at com.arm.cmsis.pack.project.RteProjectUpdater.loadRteConfiguration(RteProjectUpdater.java:219)
	at com.arm.cmsis.pack.project.RteProjectUpdater.loadConfigFile(RteProjectUpdater.java:162)
	at com.arm.cmsis.pack.project.RteProjectUpdater.runInWorkspace(RteProjectUpdater.java:105)
	at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:39)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
```

ReinhardKeil, 2016-03-17:
What would be the benefit of that?  Why is RTE a problem for you?

marek-trmac, 2016-03-17:
For example use case: I want to share my Eclipse project to my colleague, that does not have RTE package installed. He will not modify RTE configuration, only compile and debug, so he does not need RTE.

What I see as an problem, I did a legal Eclipse operation (file deletition), and ARM Eclipse plugin throws an exception during re-opening the project.

ReinhardKeil, 2016-03-17:
thanks, we will look into it.

edriouk, 2016-03-17:
The problem here is that an RTE project is dynamic and refers to location where packs are installed: 
- device information is taken from packs 
- most of headers are in pack location
- some files in RTE folder are links pointing to pack location

Just removing the RTE nature will not solve the problem. Some kind of an export procedure is required to store all required information locally and update build settings accordingly. 
While it is feasible for files and headers, there is no place to store device information in a generic C/C++ project. 

As far as deletion of an .rteconfig is concerned it is inline with Eclipse practice:
- deletion confirmation is required 
- Eclipse provides "Undo" to restore the file
- other project types also cannot tolerate file removal, for instance plugin.xml from RCP development project

marek-trmac, 2016-03-18:
For dynamic linking - would it be possible to create standalone project without external dependency (e.g. copy all files into Eclipse project)? This would be useful feature for sure.

> other project types also cannot tolerate file removal, for instance plugin.xml from RCP development project
Yes, they tolerate it correctly, e.g. does not throw an exception.