ARM-software/cmsis-pack-eclipse #18: Is pack generator supported by this plugin?

marek-trmac, 2016-03-18:
Hi,
I have a question (sorry if the issue is not right place where to ask): 
Is external generator supported by this plugin for configuration of the software? I mean this part of CMSIS specification: https://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_generators_pg.html

Are you aware of any example of such pack?

Thanks, Marek

edriouk, 2016-03-18:
Generators support (processing of gpdsc files) is not implemented the plug-in yet, but planned. 
At the time being two applications have options to generate gpdsc packs: Infineon Dave and STM32CubeMX.

marek-trmac, 2016-03-18:
Thanks for your answer. Do you have any estimation, when the feature will be supported?

edriouk, 2016-03-18:
Currently I cannot provide any estimation, since target date has not been specified yet.

edriouk, 2016-12-20:
Implemented in release 2.1.0

closed 2019-02-22 by ReinhardKeil