ARM-software/cmsis-pack-eclipse #19: URL sometimes is not displayed complete

marek-trmac, 2016-03-24:
In NPW, URL sometimes is not displayed complete. I can reproduce it this way:
1. Click on the first derivative; Observation: layout updated and URL displayed
[image]
2. Click on the last derivative (or any other derivative with longer name); Observation: URL not complete
[image]
3. Manually resize dialog; Observation: URL complete
[image]

marek-trmac, 2016-03-24:
It would be good to detect, if layout needs to be updated

edriouk, 2016-07-27:
Rejected: the proposed change interferes with layout engine and in case of a long URL completely hides the right side of the widget.
A tooltip to display the entire URL is  introduced in the 2.0 release instead.

closed 2016-07-27 by edriouk