ARM-software/cmsis-pack-eclipse #21: no processors listed in RTE project wizard

hackrid, 2016-04-27:
Hi,

please see the screenshot below.

Notice that the stm32f4 pack is installed but it's not possible to select a processor from that dialogue.

[image]

ilg-ul, 2016-04-27:
you probably need to configure again the path to the packages folder, these plug-ins do not automatically reuse the definition from the GNU ARM Eclipse Packs Manager.

hackrid, 2016-04-27:
no success

edriouk, 2016-04-27:
${cmisi_pack_root}, set by CMSIS Pack -> CMSIS Pack root folder preference

hackrid, 2016-04-27:
ahh... did not noticed there were an extra preferences entry.

Window -> Preferences -> CMSIS Pack

now it lists the processor. thank you!

closed 2016-04-27 by hackrid