ARM-software/cmsis-pack-eclipse #22: CMSIS-Core missing

hackrid, 2016-04-27:
I'm trying to create a RTE Config. But it's not straight forward.

can you tell me what I do wrong regarding the error shown in the picture below?
[image]

jkrech, 2016-04-27:
The Pack Installer shows that it has CMSIS 4.5.0 installed, however the corresponding Cclass="CMSIS" is not visible in your Components window. Either you have installed the PACK while you were already configuring and the Model/Engine needs to reread the installed PACK descriptions, or the Folder where th pack installer installed the PACKs is different from where the Model reads the description from. Have you tried a refresh already ? (F5).

edriouk, 2016-04-27:
F5 refreshes the project, not opened RTE editor. Either trigger cmsis_pack_root preference change or  close it and re-open the *.rteconfig file again.

hackrid, 2016-04-27:
ahhh.... I did some reload tricks... close the project, switch workspace, switch back, reopen project.
it works now.

thank you very much!

closed 2016-04-27 by hackrid