ARM-software/cmsis-pack-eclipse #24: Check for valid CMSIS pack root folder before attempting web update

JohnKBartholomew, 2016-08-11:
With the v2.0 plugins installed in Eclipse, but no CMSIS root folder value set (i.e. it's just empty), I get the following errors in the console output if open the CMSIS Pack Manager window and select the "Check for Pack updates on the web" button:

Parsing "http://www.keil.com/pack/index.idx"...
Error while refreshing: "null", ignored
Error while refreshing: "null", ignored
[etc]

When I set a valid location in my CMSIS Pack root folder entry, and try again, I do not see these errors.

Before the update is attempted, the CMSIS code should check for a valid root folder location - and display an error popup if it is not valid.

JohnKBartholomew, 2016-08-18:
With the 08/18/16 update, I'm no longer seeing this problem.

JohnKBartholomew, 2016-08-18:
My mistake - this was not fixed with the recent update

edriouk, 2016-12-20:
Fixed in release 2.1.0

JohnKBartholomew, 2017-08-17:
I now see a useful error message in my console window:
  Please set CMSIS Pack root folder in "Window->Preferences->CMSIS Packs" and try again

Perfect - thanks!

closed 2017-08-17 by JohnKBartholomew