ARM-software/cmsis-pack-eclipse #25: Errors in user visible strings in RefClient code

JohnKBartholomew, 2016-08-11:
Some minor modifications are needed in several  RefClient user visible strings:

RefClientPackInstaller.java line 32:
   String message = "Required GNU ARM C/C++ Cross Toolchain is not installed.\nCopy the example anyawy?";

  "anyawy" should be "anyway" above
---
 ARM.RefClient.pdsc, line 35:
      <description>ARM CortexM-4 example demonstarining PackInstaller functionality to install examples. Note: This example requires Cross ARM GCC toolchain to be installed</description>

  Both "Cortex-M4" and "demonstrating" are incorrect above.

edriouk, 2016-08-16:
The typos are fixed and pushed to the repository.

JohnKBartholomew, 2016-08-16:
Verified typos are fixed

closed 2016-08-16 by JohnKBartholomew