ARM-software/cmsis-pack-eclipse #27: NPE i CpPackManager if no pack installer has been set

jesperes, 2016-08-29:
Hi,

In upgrading from 1.0 to 2.0.1, I discovered that CpPackManager assumes that there is a pack installer set, and will NPE (e.g. by calling fPackInstaller.stopPackWatchThread()) if it isn't.

Is there a list of known incompatibilities between 1.x and 2.x? Some sort of migration guide would be nice.

edriouk, 2016-08-30:
Thank you for the pull request anyway. Together with some additional changes it will help isolate pack installer functionality from the rest of plug-ins.  
However, are you considering to use this new functionality provided with version 2.0? 
You should also have a look at ICpEnvironmentProvider interface that could make plug-in integration in your environment easier. 

As far as incompatibilities are concerned, they are mostly related to moving **IBuildSettings** and **IMemorySettings** interfaces from  **com.arm.cmsis.pack.build.setttings** to **com.arm.cmsis.pack** that should be not relevant for you.

jesperes, 2016-08-30:
Yes, we will look into the pack installer functionality.

I noticed that the IBuildSettings has moved, which actually does affect us as we also have a toolchain adapter plugin for our compiler. We're in the process of updating them to 2.0.1 as well.

edriouk, 2016-12-20:
Fixed in release 2.1.0

closed 2019-02-22 by ReinhardKeil