ARM-software/cmsis-pack-eclipse #31: AdvisedCellLabelProvider adds space to labels

jesperes, 2016-08-30:
The class AdvisedCellLabelProvider has a "workaround" which looks like this:

```
			String s = columnAdvisor.getString(element, index);
			if(s != null && !s.isEmpty()) {
			// workaround: append space to the text to avoid last character truncation the cells
				s += ' ';
			}
			cell.setText(s);
```

This workaround causes widespread breakage in out SWTBot tests, which presumes that the cell text is unmodified.

The workaround feels very kludgy, maybe there is a better way to make JFace measure the text extent correctly?

edriouk, 2016-08-30:
We need to investigate why it has been done. As far as I remember it was not the extent calculation, but a JFace drawing problem on one of the platforms. The very last letter in a cell was truncated in the middle even if the cell width was wide enough, for instance R you could be seen as I.

Could you modify your tests to truncate trailing spaces?

jesperes, 2016-08-30:
I have disabled the workaround in our local fork, so this is not a critical issue at the moment.

edriouk, 2016-12-20:
We have had the similar problems with SWTBot and solved the problem by modifying tests.