ARM-software/cmsis-pack-eclipse #32: Make linker option method non-private to allow link option overriding in toolchain adapter

JohnKBartholomew, 2016-10-10:
I'm working on a toolchain adapter which will read in my options values (using the <environment> section <LMisc> tag) and it works - except I cannot access all the existing default linker options to add to mine.

For example, in the code snippet below, I modeled my code after what I found in ArmGccToolchainAdapter.updateLinkerMiscOption() - I can call getFpuOptionValue() as it is public, but I cannot call getFloatAbiOptionValue() because it is private. Please consider making this second method public or at least protected so I can access it to build my enhanced LMisc based linker option string.

		case  IBuildSettings.LMISC_OPTION:
			sb = getMiscOptionValue(sb, IBuildSettings.RTE_LMISC, option, buildSettings);
			
			// Code below replaces parent ArmGccToolchainAdpater call:
			//   updateLinkerMiscOption(configuration, tool, option, buildSettings);
			// This is because this call would not allow us to include our settings above
			// We could not include the call to getFloatAbiOptionValue() however as this
			// method is private in the ARM CMSIS-Pack Mgmt plugins v2.0.1
			
			String fpu = getFpuOptionValue(buildSettings);
			if (fpu != null && !fpu.isEmpty()) {
				sb = sb.append(SPACE);
				sb = sb.append(MFPU);
				sb = sb.append(EQUAL);
				sb = sb.append(fpu);
			}
			
			// Not part of public or protected interface...!
			//String floatAbi = getFloatAbiOptionValue(buildSettings);
			//if (floatAbi != null && !floatAbi.isEmpty()) {
			//	value += SPACE + MFABI + EQUAL + floatAbi;
			//}
			
			ManagedBuildManager.setOption(configuration, tool, option, sb.toString());
			break;

edriouk, 2016-10-11:
Hi John,
Thank you for reporting that. I have overlooked the private access at this place. 
I'll change it to public in the next release.

edriouk, 2016-12-20:
Implemented in release 2.1.0

JohnKBartholomew, 2017-08-17:
Verified as fixed (in version 2.2.0). Thanks!

closed 2017-08-17 by JohnKBartholomew