ARM-software/cmsis-pack-eclipse #34: 403 when using Eclipse NEON

thirtytwobits, 2016-11-18:
I get 403 responses when using Eclipse Neon to access the pack URL http://www.keil.com/pack/index.idx (default address). I can open the xml from the same host in a browser. Not sure what's going on to be honest.

edriouk, 2016-11-21:
Does it works using Eclipse Mars from the same computer?
What OS do you use?

thirtytwobits, 2016-11-21:
I don't have Eclipse Mars on my computer. I use OSX.

ilg-ul, 2016-11-22:
> I don't have Eclipse Mars on my computer.

@thirtytwobits, you can have as many Eclipses as needed on a single computer. It's just a matter of downloading a zip and unpacking it in a folder of your choice. it does not interfere at all with any other Eclipses. just be sure you use separate workspaces.

edriouk, 2016-12-02:
We have tested the plug-in version 2.0.1 with Eclipse Neon at Windows and Mac and could not reproduce the issue.  Does it still persists in your environment? 
Mac OS details: OS X EI Capitan  System Version: OS X 10.11.6 (15G31)  Kernel Version: Darwin 15.6.0

thirtytwobits, 2016-12-03:
Yep. The problem seems to have fixed itself. It may have been something to do with our corp net as I'm able to access from home okay. I can't actually install any of the packs yet but that's a separate problem I'll investigate later.
Sorry for the noise but thanks for the due diligence.

closed 2016-12-03 by thirtytwobits