ARM-software/cmsis-pack-eclipse #37: Keyword

mzamaron, 2017-01-17:
In CMSIS a couple of keywords are proposes:
Board Support,  CMSIS, Device, CMSIS Driver 
File System, Graphics, Network, USB for cClass attribute.
CORE, DSP, RTOS for cGroup attribute.
I would like to know if it's possible to deploy standard keywords across partners/vendors to easily match the conditions on external packs components for Group, Variant, bundle,..
The list of keywords must be accessible on the Keil.com web site.

jkrech, 2017-01-20:
The best place for this agreed names for 
Cclass and Cgroup as well as standard Bundle or Variant names or naming conventions is the CMSIS-Pack specification. http://arm-software.github.io/CMSIS_5/Pack/html/index.html

Please provide suggestions for names you see need for.

ReinhardKeil, 2017-02-07:
duplicate - please post only once.

closed 2017-02-07 by ReinhardKeil