ARM-software/cmsis-pack-eclipse #41: How to make examples show up in the examples tab?

engsergiosoares, 2017-07-14:
Hi guys, thank you for your great work with this plugin. It makes our lives easier.
Sorry if this is a dumb question but I couldn't find an answer yet.

I have a pack I built for Atollic TrueSTUDIO and by just using
<project>
        <environment name="**atollic**" load=".project"/>
</project>

I can get my examples listed in the examples tab of the pack manager. 
I tried the packs plugin with Eclipse Oxygen and it seems to work fine. I can install my packs, create RTE projects, etc. Unfortunately, my example projects don't show up in the Examples tab of the CMSIS Pack Manager. I've tried setting the environment name to "eclipse" in the .pdsc file but it didn't work. Is there any other name I should use to make this work on pure eclipse? 

Regards,
Sergio Soares

edriouk, 2017-07-17:
Hi Sergio,

Please note that the plug-in has a limited usage in a generic Eclipse CDT.
Its primary goal to allow IDE developers to enhance their tools with CMSIS pack support.
Since such environments may significantly differ, in particular in debugger integration, the plug-im must be sure the example is sutable for that environment.

Therefore to see your examples in the list you need to supply a plug-in with your implementation of ICpEnvironmentProvider interface by extending EnvironmentProvider extension point.  In that  implementation you override isExampleSupported() method to return true for your examples.
Please look at com.arm.cmsis.pack.refclient plug-in for an example. 

Best regards,
Evgueni

engsergiosoares, 2017-07-20:
Thank you, Evgueni.

closed 2017-07-20 by engsergiosoares