ARM-software/cmsis-pack-eclipse #42: Headless build with cmsis-pack-eclipse?

engsergiosoares, 2017-07-28:
Hi guys,

Thank you for the hard work with the plugin!

I'm trying to do a headless build with Eclipse and an RTE C++ project but I'm having some difficulties (not sure if I'm doing something wrong or if the plugin simply doesn't support it).
It seems the cmsis-pack-eclipse plugin is not being invoked to resolve the include paths, so the headless build uses the paths contained in the .cproject, without updating them. The problem is whenever I update the pack version, all the include paths are invalid again. 
This issue doesn't happen when using the GUI, since the packs plugin seems to automatically detect the new version of the pack and update the .cproject file accordingly.

Is there a work around to make it work in a headless build? Like copying plugin configuration files somewhere...?

Thank you!
Best regards,
Sergio Soares
Software Tools Developer - ON Semiconductor

edriouk, 2017-07-29:
Hi Sergio,
thank you for reporting this. We have not tested a headless build, so we are unware of the problem. 
Triggering a project update was a tricky problem to solve even for GUI. CDT does not provide a necessary hook or an extension point to be called during .cproject load. In GUI mode CMSIS plug-in simply monitors for a certain CDT job to be triggered. According to your observations  this does not work in headless mode.
We need to look at this, but I cannot promise any solution in near future.
Best regards,
Evgueni  Driouk

engsergiosoares, 2017-07-31:
OK, no problem, Evgueni. I just wanted to confirm if I wasn't doing something wrong before deciding to go with a more "brute-force-like" solution.
We'll write a script to update the paths before performing the headless builds on our build machines (this should solve our problem for now).

Thank you!
Regards,
Sergio.

engsergiosoares, 2019-01-02:
Hi @edriouk: do you have any plans to add headless build support for this plugin?
We would love to have this feature to use it for automated builds. 
If there's anything we can do to help, please let us know.

Regards,
Sergio

edriouk, 2019-01-04:
The hadless build support is added to the recently published 2.4.0 release. 
Run eclipsec.exe (Windows) or exlipce (Linux) as follows: 
eclipsec.exe -nosplash -application com.arm.cmsis.pack.project.CmsisHeadlessBuilder  -help
to get the full list of command line options

engsergiosoares, 2019-01-04:
Great news! Thank you very much, @edriouk .

closed 2019-01-04 by engsergiosoares

engsergiosoares, 2019-04-11:
Hi @edriouk ,

The plugin v2.4.0 works fine on GUI mode. But I tried running the headless build help command and I'm getting this error:
"java.lang.RuntimeException: Application "com.arm.cmsis.pack.project.CmsisHeadlessBuilder" could not be found in the registry."

Please find the full log attached
[1555011646408.log](https://github.com/ARM-software/cmsis-pack-eclipse/files/3070307/1555011646408.log)
.

Am I missing any file?

Thanks,
Sergio.

edriouk, 2019-04-12:
Hi Sergio,

you need to specify the application id in the command line instead of the class name:

-application **com.arm.cmsis.pack.project.headlessbuild**

Please try.
Evgueni

engsergiosoares, 2019-04-16:
Thanks, Evgueni. Worked for me!