ARM-software/cmsis-pack-eclipse #43: Preserving library linking order?

engsergiosoares, 2017-08-01:
Hi guys,

I'm playing with the refclient example and the gnuarmeclipse plugin. I'm trying to incorporate them in our eclipse-based IDE. Almost everything works out of the box, which is great!!!

In our pack, we have pre-built libraries (.a files), that need to be linked in a specific order. The problem is that the plugin re-orders the library and updates the .cproject file in alphabetical order, which unfortunately doesn't work for us. I've also noticed that the plugin sets the libraries as "other objects", instead of libraries (with its library paths).

Can you guys give me a hint on where I could modify the plugin to preserve a certain library order specified by the user?

Regards,
Sergio Soares
Software Tools Developer - ON Semiconductor

edriouk, 2017-08-28:
Hi Sergio,
The CMSIS projects are designed to work with components, not single files. 
The components can contain libraries and depend on other components that also have libraries and even refer to the same libraries. Therfore there is no way to preserve a defined order of libraries.
It is a well known problem of a GCC linker that evaluates libraries for unresolved externals only once.
One possibility to resolve that is to add all libraries twice. Thus you can modify a toolchain adapter to do that for you. 

As far as using "other objects" instead ogf "libraries": that is the only way to supply libraries with absolute paths to GCC linker. Using separate "libraries" and "library paths" options has a potential problem of library conflicts when several components containg different libraries with the same names.

engsergiosoares, 2019-05-06:
Hi edriouk, thanks for the help here.

We are using the gnuarmeclipse toolchain adapter and I overrode the _setStringListOptionValue_ method from the super class _RteToolChainAdapter_ in _GnuarmeclipseToolChainAdapter_ to duplicate the entries in 'arrayValue'. Seems to work for us to resolve the GCC linker single pass to resolve externals.

Is this the best place to apply this change? Or would you suggest somewhere else?

If you would like me to send a pull request, let me know (not sure this is of interest of other people).

Regards,
Sergio.

closed 2019-05-06 by engsergiosoares

edriouk, 2019-05-07:
Hi Sergio,

Please submit the pull request. I'll review it. 

Best regards,
Evgueni

engsergiosoares, 2019-05-16:
Hi Evgueni,

After some more investigation, we found out there is an option in the GCC linker (-Wl,--start-group ${OBJECTS} -Wl,--end-group) that enables it to resolve circular dependencies for a group of objects/libs in multiple passes (until all externals are resolved). So, we just had to add it in the command line for the Linker Expert Settings.
 
This article explains it in details: https://eli.thegreenplace.net/2013/07/09/library-order-in-static-linking

I'll close this issue. Thanks for the help.

edriouk, 2019-05-16:
Hi Sergio,

Thank you for the information.

Best regards,
Evgueni