ARM-software/cmsis-pack-eclipse #45: CMSIS Configuration Wizard not show GUI editor

nash900, 2017-09-05:
When open source file (Code Example) with "Configuration Wizard" in Eclipse, only see "Source Editor" tab but not see CMSIS GUI tab. 
Open the same file in uVision IDE, it's ok.

Env.:
* Eclipese version: eclipse-cpp-oxygen-R-win32-x86_64.zip
* CMSIS-pack-eclipse version: CmsisPackPlugIn2_2_0.zip

Anything am I missing? Thanks.

BTW, the Code Example is found in https://www.keil.com/pack/doc/CMSIS/Pack/html/configWizard.html:
//-------- <<< Use Configuration Wizard in Context Menu>>> -----------------
//
// <h>Thread Configuration
// =======================
//
//   <o>Number of concurrent running threads <0-250>
//   <i> Defines max. number of threads that will run at the same time.
//   <i> Default: 6
#ifndef OS_TASKCNT
 #define OS_TASKCNT     6
#endif

// snip

// </h>
//------------- <<< end of configuration section>>> -----------------------

edriouk, 2017-09-05:
Thnak you for reporting this issue.
The problem is that the config parser in the plug-in is too strict in matching open and end tags. The tags are hard coded in the plug-ing and contain exactly one space before >>>, i.e.
//-------- <<< Use Configuration Wizard in Context Menu >>> -----------------
//------------- <<< end of configuration section >>> -----------------------
Please change those tags and re-open the file in Config Wizard.

We will fix the logic in the next release.

nash900, 2017-09-05:
Edriouk, thanks a lot for your quick reply. Perfect! It's working now.

edriouk, 2017-11-22:
I believe this issue is fixed in 2.3.0 release. The ConfigWizard parser should ignore the spaces
Evgueni

ReinhardKeil, 2019-02-22:
Closing as no further feedback.

closed 2019-02-22 by ReinhardKeil