ARM-software/cmsis-pack-eclipse #46: CMSIS Pack / Eclipse improperly installs pack file...

forehead2k, 2017-09-27:
Eclipse (Neon.3, running on macOS Sierra)
CMSIS Pack Eclipse Plugins (v2.2.0.20170326_1123130)
... and ...
uVision (v5.24.2.0, on Windows 10)

When installing a CMSIS Pack file on Windows in uVision, it honors the full version in the PDSC "release" XML. E.g.,

Cypress.PDL.3.1.pack contains the following <release> in its PDSC file:
        <release version="3.1.0+build.624" date="2017-08-24">

uVision installs to:
<PACK_ROOT>\Cypress\PDL\3.1.0+build.624\...

CMSIS Eclipse plugin installs to:
<PACK_ROOT>/Cypress/PDL/3.1/

When uninstalling the pack file, it looks for:
<PACK_ROOT>/Cypress/PDL/3.1.0+build.642/

Which does not exist, and uninstall fails. 

Note this is being done from a custom Eclipse-derived tool that integrates the CMSIS Eclipse plugin via a direct call in to the plugin. E.g., (error checking redacted for simplicity):

        final ICpPackManager packMgr = CpPlugIn.getPackManager();
        final ICpPack pack = packMgr.getPacks().getPack(packId);
        final ICpPackInstaller packInstaller = packMgr.getPackInstaller();
        packInstaller.removePack(pack, true);

The ICpPack is correctly found / references the pack we wish to uninstall.

forehead2k, 2017-09-27:
PS- The Eclipse CMSIS Pack Manager is also unable to uninstall the pack file. 

- It's removed from the UI.
- The files are not removed from disk.
- Restarting Eclipse causes the offending pack file to re-appear in the UI.
- If you turn on Eclipse "-consoleLog" the uninstall process generates the following stack trace:

org.eclipse.swt.SWTException: Failed to execute runnable (java.util.NoSuchElementException)
	at org.eclipse.swt.SWT.error(SWT.java:4533)
	at org.eclipse.swt.SWT.error(SWT.java:4448)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:185)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4032)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3708)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:1121)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1022)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:150)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:693)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:138)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
Caused by: java.util.NoSuchElementException
	at java.util.TreeMap$PrivateEntryIterator.nextEntry(TreeMap.java:1209)
	at java.util.TreeMap$EntryIterator.next(TreeMap.java:1247)
	at java.util.TreeMap$EntryIterator.next(TreeMap.java:1242)
	at com.arm.cmsis.pack.item.CmsisMapItem.getFirstChild(CmsisMapItem.java:126)
	at com.arm.cmsis.pack.item.CmsisMapItem.getFirstChild(CmsisMapItem.java:1)
	at com.arm.cmsis.pack.item.CmsisTreeItem.getFirstChild(CmsisTreeItem.java:1)
	at com.arm.cmsis.pack.installer.ui.views.DevicesView.stopAtCurrentLevel(DevicesView.java:57)
	at com.arm.cmsis.pack.installer.ui.views.DevicesView$DeviceViewContentProvider.hasChildren(DevicesView.java:91)
	at org.eclipse.jface.viewers.AbstractTreeViewer.isExpandable(AbstractTreeViewer.java:2136)
	at org.eclipse.jface.viewers.TreeViewer.isExpandable(TreeViewer.java:538)
	at org.eclipse.jface.viewers.AbstractTreeViewer.isExpandable(AbstractTreeViewer.java:2166)
	at org.eclipse.jface.viewers.AbstractTreeViewer.updatePlus(AbstractTreeViewer.java:2849)
	at org.eclipse.jface.viewers.TreeViewer.updatePlus(TreeViewer.java:786)
	at org.eclipse.jface.viewers.AbstractTreeViewer.updateChildren(AbstractTreeViewer.java:2741)
	at org.eclipse.jface.viewers.AbstractTreeViewer.internalRefreshStruct(AbstractTreeViewer.java:1913)
	at org.eclipse.jface.viewers.TreeViewer.internalRefreshStruct(TreeViewer.java:669)
	at org.eclipse.jface.viewers.AbstractTreeViewer.internalRefresh(AbstractTreeViewer.java:1888)
	at org.eclipse.jface.viewers.AbstractTreeViewer.internalRefresh(AbstractTreeViewer.java:1845)
	at org.eclipse.jface.viewers.AbstractTreeViewer.internalRefresh(AbstractTreeViewer.java:1831)
	at org.eclipse.jface.viewers.StructuredViewer.lambda$0(StructuredViewer.java:1519)
	at org.eclipse.jface.viewers.StructuredViewer.preservingSelection(StructuredViewer.java:1458)
	at org.eclipse.jface.viewers.TreeViewer.preservingSelection(TreeViewer.java:354)
	at org.eclipse.jface.viewers.StructuredViewer.preservingSelection(StructuredViewer.java:1419)
	at org.eclipse.jface.viewers.StructuredViewer.refresh(StructuredViewer.java:1519)
	at org.eclipse.jface.viewers.ColumnViewer.refresh(ColumnViewer.java:523)
	at org.eclipse.ui.dialogs.FilteredTree$NotifyingTreeViewer.refresh(FilteredTree.java:1240)
	at org.eclipse.jface.viewers.StructuredViewer.refresh(StructuredViewer.java:1478)
	at org.eclipse.ui.dialogs.FilteredTree$NotifyingTreeViewer.refresh(FilteredTree.java:1228)
	at com.arm.cmsis.pack.installer.ui.views.PackInstallerView.handleRteEvent(PackInstallerView.java:503)
	at com.arm.cmsis.pack.installer.ui.views.PackInstallerView.lambda$3(PackInstallerView.java:478)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:35)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:182)
	... 23 more

forehead2k, 2017-09-27:
Let me know if you need anymore information.

edriouk, 2017-09-29:
Thnak you for reporting this issue. Obviously we have not tested versions with metadata (after +).
In fact, both uVision and and Eclipse plug-ins work incorrectly.
Accordin to semantic versioning in case of version 3.1.0+build.624, the meta-data should be dropped and the following directory should be used consistently acroos the tools: 
<PACK_ROOT>/Cypress/PDL/3.1.0
all operations over packs should also respect that.

We will fix the ill behavior in the next version, however i cannot promise you any timeframe.

forehead2k, 2017-09-29:
Much appreciated. I realize you can’t commit to any specific time frame, but could you indicate whether it was at all likely to happen before the end of the year?

Thx,
Nick

This message and any attachments may contain confidential information from Cypress or its subsidiaries. If it has been received in error, please advise the sender and immediately delete this message.

edriouk, 2017-11-07:
Hi Nick,
Sorry for a delay. I have eventually came to fixing this issue. 
However I have noticed that in you case the pack filename does not conform to CMSIS standard, see
http://www.keil.com/pack/doc/CMSIS/Pack/html/cp_Packs.html#PackFilenames
In you case the correct filename should be Cypress.PDL.3.1.0+build.642.pack 
Then the Eclipse pack manager also correctly installs and uninstalls the pack.

Nevertheless I still need to fix the issue that the pack should be installed in <PACK_ROOT>/Cypress/PDL/3.1.0 directory, not the <PACK_ROOT>/Cypress/PDL/3.1.0+build.642

edriouk, 2017-11-22:
Hi Nick
I believe this issue is fixed in 2.3.0 release. The pack should be installed in a path without meta information.
Could you try if it works correctly. Please let me know if you still experience exceptions during install/uninstall operations.
Evgueni

ShuopengDeng, 2018-01-12:
If
I am using 2.3.0 release Pack filtering still fails for inconsistent versions. In the .rteconfig file, if I have a pack install under <PACK_ROOT>/Cypress/PDL/3.1.0 and the release version number is 3.1.0+build473 in PDSC, then rteconfig file's Packs' tab's UI displays the version 3.1.0+build.474, but when when it tries to find the corresponding pack for fitering, it fails due to inconsistent version numbers. (See attached images).

[image]

[image]

ShuopengDeng, 2018-01-12:
Meant to attached the error image instead of a duplicated image in the previous comment.
[image]

edriouk, 2018-02-21:
I'm closing the issue since related #58 is closed

closed 2018-02-21 by edriouk