ARM-software/cmsis-pack-eclipse #55: CMSIS pack not seen in Window->Preferences tab in Eclipse Mars

radhika-raghavendran, 2018-01-15:
I am using Eclipse Mars 2 Release (4.5.2) Build ID: 20160218-0600
The CMSIS pack management plugin version being used is CMSIS-Pack Eclipse Plug-in 2.3.1 
After installation of the plugin, on restart of eclipse, I cannot see the CMSIS pack root folder configuration tab in Window->Preferences.
Can you please tell me the latest eclipse version and CMSIS plugin to use? I did see the readme and the issues logged but could not find much information.

Thanks
Radhika

edriouk, 2018-01-15:
Hi Radhika,
The latest plug-in version is 2.3.1, it has been tested on Eclipse Mars 4.5.2 (on Windows).
Please try to re-install the plug-in:
- uninstall the plug-in, restart Eclipse and exit it. 
- go to plugins directory of your Eclipse and delete all com.arm.cmsis.*.jar files (if any)
- start Eclipse againg with a fresh workspace and try to install the plug-in again
Best regards,
Evgueni

radhika-raghavendran, 2018-01-15:
Thank you Evgueni.
I was able to see the CMSIS pack once I uninstalled and reinstalled the plug-in.
Now, I have a problem of not able to view the examples created in the CMSIS-pack. I did install the refpack as mentioned in point 5 in the readme (**5.(Optional) Install supplied ARM.RefClient.0.2.0.pack to evaluate example importing functionality**)
I still do not see the examples in the CMSIS pack view. 
[image]

edriouk, 2018-01-15:
Hi Radhika,
The CMSIS plug-in purpose is to bring CMSIS-pack support to development tools that provide other important functionality: toolchain integration, flash download, debug and trace. Therefore the primary audince for the plug-ins are engineeres developing the tools. RefClient plu-ing is just a reference example showing how to integrate CMSIS plug-ins ito particular tool.
As far as I undestand you use plain Eclipse with CMSIS plug-ins installed. That stand-alone configuration has very limited usage: pack management (install, update, delete) and basic CDT project management. 
It does not know what to do with pack examples. For the selected TI pack contains exaples that are only available for ARM MDK (aka uVision), IAR and Atollic True Studio environments. 
Therfore your plain Eclipse environment does not show any example.

radhika-raghavendran, 2018-01-17:
Hi Evgueni,
We are creating a CMSIS pack for our microcontroller. We include drivers, boards and examples in this pack. I created an example eclipse project and included it in our CMSIS pack.  On installation of the pack using CMSIS pack manager in Eclipse, I still do not see the example. In principle, I should be seeing the example folder. Also, is the CMSIS pack manager plugin supported for the latest version of Eclipse (Oxygen)?

Thank & Regards,
Radhika

edriouk, 2018-01-17:
Hi Radhika,

Which development tools support your examples : Keil, IAR, Atollic, other? 
Each of those development tools show only their examples. 
There are plans to provide generic tool-agnostic example framework in CMSIS, but this functionality is still at initial development stage => will not be available in the near future. 

CMSIS Pack plug-in can be used with Eclipse Oxygen. 

Best regards,
Evgueni

radhika-raghavendran, 2018-01-18:
Example was made using the Eclipse IDE. The same example is bundled in the CMSIS pack for distribution. On installing the CMSIS pack I expect to see the example in Eclipse IDE but I don't.
Attached is a snapshot. The .project and .cproject are also present in the pack structure.
[image]
[image]

edriouk, 2018-01-18:
Again, 
What plain Eclipse IDE could do with your example:
- what compiler toolchain should be used to built the example?
- what tool should perform flash download and how?
- what debugger is supposed to be used?

Please have for instance to an example description  in pack:
`    <example name="BlinkLED" folder="Examples/BlinkLED" doc="Abstract.txt" version="3.2.0">
            <description>This is a basic example demonstrating the development flow and making the LED on the board blink</description>
            <board name="MSP-EXP432E401Y" vendor="TexasInstruments"/>
            <project>
                <environment name="uv"  load="ARM/BlinkLED.uvprojx"/>
                <environment name="iar" load="IAR/BlinkLED.ewp" />
                <environment name="atollic" load="Atollic/.project" />
            </project>
        </example>
    `
Note `environnment` elements describing supported development tools. Those elements are used by the corresponding tools to check if an example is supported. To do that those tools must override  ICpEnvironmentProvider.getName() method to return the environment name. 

The default implementation ICpEnvironmentProvider.getName() returns an empty string signalling that none of the examples is supported.

radhika-raghavendran, 2018-01-19:
The compiler toolchain in plain eclipse is set to ARM-NONE-EABI-GCC and the tool for flash download is Segger JLink using the ARM GDB debugger, which is all set while the project is created in Eclipse IDE.

The .pdsc file also contains the example element along with all the required sub-elements. Here is what the example for eclipse project looks like.
`		<example name="EC-GPIO" doc="" folder="Boards/NEBO-DUE/Examples/eclipse-gpio">
<description>NEBO-DUE board GPIO driver example on Eclipse</description>
<board name="NEBO-DUE" vendor="ONSemiconductor"/>
<project>
<environment name="eclipse" load="Eclipse/.project"/>
</project>
<attributes>
<component Cclass="CMSIS"     		Cgroup="CORE"/>
<component Cclass="Device"    		Cgroup="Startup"/>
<component Cclass="Device"    		Cgroup="NEBO drivers"/>
<component Cclass="Application"    	Cgroup="NEBO-DUE" Csub="GPIO"/>
</attributes>
</example>`
 

> The default implementation ICpEnvironmentProvider.getName() returns an empty string signalling that none of the examples is supported.

I did not completely understand this point. Do I need to make some changes in the plugin for it to recognize the eclipse IDE name?

edriouk, 2018-01-19:
Yes, you have to do some changes in the plug-in and deliver it to the customers together with all pieces that are needed to run your example :
- ARM GCC toolchain 
- plug-in integrating the toolchain to Eclipse (I asume that is one from gnuarmeclipse)
- Segger Jlink integration to Eclipse 
- ARM GDB integration to Eclipse 

Alternatively you can provide instructions how to install required plug-ins and how to import your project or even better to make your example work for other development environments: Keil, IAR, Atollic. 
Please note that for the time being only Atollic is Eclipse-based.

radhika-raghavendran, 2018-01-23:
If possible can you please provide me an example of how the plugin needs to be modified?  Since Atollic is an Eclipse-based IDE, some modifications would have been made for Atollic to recognize the project and launch it. I would much appreciate it if you can give me some information on this.
Also, I am using the plugin from this website and installing the plugin as an archive file in bare Eclipse. I followed the directions given in Readme.md file in your GITHUB repository landing page.

Thanks
Radhika

radhika-raghavendran, 2018-01-30:
I am currently using the RefClient environment for my examples and it does show up on Eclipse. Please let me know whenever this repository updates to supporting "Eclipse" environment with the need to use RefClient plugin. We would like to use that plugin and update our examples in our CMSIS pack distribution. 
Thank you for the help.

closed 2018-01-30 by radhika-raghavendran

edriouk, 2018-01-30:
Hi,
RefClient is just a referece for tool developers, it is not included in the plug-in and you should not rely on its presence in the customer's environment.
What I can consider is to provide functionality to see all examples available in the packs regardlesss if they are supported by current environment with the possibiliry to install those that have .project file in it (without guarantee the example works or even compiles).
However I cannot commit to any timeframe this functionality comes in.

satishchandra-mindtree, 2018-05-28:
Hi Radhika,

Were you able to get the examples visible? Could you mention how you got the examples to be visible in standalone eclipse.I am using CMSIS-Pack Eclipse Plug-in 2.3.2 and its not visible for me with my own environment name. 

Thanks,
Satish.

vaughanatworld, 2018-09-29:
I'm running "CMSIS Pack Eclipse plug-ins	2.3.2.20180410_085642" and examples are not visible (as above).
Has there been any progress since April?
Thanks.
Brian

radhika-raghavendran, 2018-10-01:
There is no progress since January.  I do not know if there is a plan to support this in the future.