ARM-software/cmsis-pack-eclipse #56: Does this support command line to import packs ?

Hoohaha, 2018-01-17:
Hi sirs，

I work with ds-mdk & iar.
There are bulit-in cmsis-pack-eclipse tool.

How could i import packs by command line?
And where the location of packs is installed on linux?

edriouk, 2018-01-17:
Hi,

There is no command-line pack management tool for Linux. On Windows you can use PackUnzip.exe and PackInstaller.exe from MDK ARM (http://www2.keil.com/mdk5/), those executables do dot require license.

You define the pack location via Eclipse preferences:  Window->Preferences->CMSIS Packs

What do you mean under pack import : 
 - download and install an available pack ?
 - unpack locally available pack?

jkrech, 2018-01-17:
The eclipse plug-ins only support the pack installation using the GUI.
Since the *.pack files are regular zip archives you can use unzip to extract the archive into a folder.
In order to avoid conflicts with the pack manager GUI plug-in you should follow the folder naming convention. A pack gets extracted into the folder named <vendor>/<pack name>/<pack version> where version is limited to <major>.<minor>.<patch>[-...] (+... is ignored).

Hoohaha, 2018-02-02:
In my situation, a lot of local packs need to be installed to IDE,  so we want to using to script to achieve athe automated installation. As far as i know ARM MDK provide a friednly tool named "PackUnzip.exe", but for cmsis-pack-eclipse we have no idea. @jkrech i think you advice is a good solution.

marguzmar, 2019-09-23:
Question was answered.

closed 2019-09-23 by marguzmar