ARM-software/cmsis-pack-eclipse #57: Support for template file creation

jesperes, 2018-01-17:
I would like to be able to detect when the user creates a template file in a CMSIS pack project. I'm trying to keep an external project file in sync with the CMSIS pack project in Eclipse, but there is no way of knowing when the user creates a template, nor what the template is called.

I'm going to attempt to implement this by triggering an RteEvent when a template file is created.

edriouk, 2018-01-17:
Do you mean New->Files from CMSIS Template?
In my opinion that is not different from just creating a new file or importing an existing one and can be monitored using IResourceChangeListener.

jesperes, 2018-01-17:
Ok, thanks. I was hoping to be able to distinguish template files. IAR Embedded Workbench users are accustomed to manually adding files to the project, and not to have them automatically appear.

After some further discussing my colleagues, I think will duck this issue for now. We need to figure out a little more in detail how to handle the case of files appearing/disappearing in the project and how this should be synced with the EW project file.

closed 2018-01-17 by jesperes