ARM-software/cmsis-pack-eclipse #58: Pack filtering does not work if Pack version does not match pack id

ShuopengDeng, 2018-02-08:
Support I install a Pack with ID "Cypress.PDL.3.1.0" and version "Cypress.PDL.3.1.0+build657", then RTE config file will not filter the pack for a fixed version.

CpPackFilter.passes(ICpPack pack) uses the ID of the pack (Cypress.PDL.3.1.0) to perform the actual filtering
RtePackCollection.createPackFiter() (and a few other places) uses the version of the pack create the pack filter.

I would recommend doing something similar the changing the method CpPackFilter.passes(ICpPack pack) to:

	@Override
	public boolean passes(ICpPack pack) {
		if(pack == null)
			return false;
		return passes(pack.getPackFamilyId() + "." + pack.getVersion());
	}

edriouk, 2018-02-08:
The described behaviour is correct as it follows Semantic Versioning Specification (https://semver.org/):

10. Build metadata MAY be denoted by appending a plus sign and a series of dot separated identifiers immediately following the patch or pre-release version. Identifiers MUST comprise only ASCII alphanumerics and hyphen [0-9A-Za-z-]. Identifiers MUST NOT be empty. Build metadata SHOULD be ignored when determining version precedence. Thus two versions that differ only in the build metadata, have the same precedence. Examples: 1.0.0-alpha+001, 1.0.0+20130313144700, 1.0.0-beta+exp.sha.5114f85.

Therefore "Cypress.PDL.3.1.0" and "Cypress.PDL.3.1.0+build657" are considered identical. In case you need to distinguish the packs during development use '-' sign instead of '+'.

ShuopengDeng, 2018-02-13:
The problem is that RTE filtering code does NOT consider "Cypress.PDL.3.1.0" and "Cypress.PDL.3.1.0+build657" to be identical. The RTE filtering code uses the version number with the build number in some places and the version number without the build number in other places. As a result, if the pack is located in a directory called "Cypress/PDL/3.1.0" but if the pdsc contains the line "<release version="3.1.0+build.675" date="2018-02-10">", the pack filtering code will fail as it considers the version 
"Cypress.PDL.3.1.0" and "Cypress.PDL.3.1.0+build657" to be different packs.

Specifically, in the RTE config file's "Packs" tab, if I try to filtering the pack "Cypress.PDL.3.1.0+build657" to be fixed version (see image packs_tab) then the RTE config file store the version WITH the build number to be to kept. But the filtering code in CpPackFilter iterate through the versions WITHOUT the number string-compare them to the store version numbers. It sees that the string "Cypress.PDL.3.1.0" and the string "Cypress.PDL.3.1.0+build657" are not equal, as it result it marks the pack "Cypress.PDL.3.1.0" to be excluded.

[image]
[image]

edriouk, 2018-02-14:
You are right, thank you for reporting that. 
I'll look for all cases where the pack version is used and make the use consistent.
However I cannot promise you a quick fix.

edriouk, 2018-02-14:
The fix is pushed to the repository it appeared to be simpler than expected. .
I have also updated 2.3.1 release by uploading pre-built plug-ins binary containing the fix.
Could ypou please test it.

ShuopengDeng, 2018-02-14:
Still broken.

When the RTE config file is first opened, the pack filtering works correctly. But if any changes to the "packs" tab are made, then the code in RtePackCollection.createPackFiler() still set the pack version to the version including the build number. This cause the pack to still get filtered out.

edriouk, 2018-02-15:
Please test the new update

closed 2018-02-15 by ShuopengDeng