ARM-software/cmsis-pack-eclipse #59: Do not invent device URL or add a capability for vendor to specify the URL

oakm, 2018-02-23:
When I open  .rteconfig Device tab, there is URL field that has a value in format:  http://www.keil.com/dd2/<vendor>/<device>/. For example: http://www.keil.com/dd2/cypress/cy8c6347bzi_bld53/
[image]

How do I change this?

edriouk, 2018-02-28:
That URL gets created when you publish a pack.  You can use `<book>` to insert your own device-related URLs. The books are seen in the same Device tab in the dedicated list.

oakm, 2018-03-01:
Thanks for the prompt response. `<book>` allows me to add a new URL. However I still cannot get rid of auto-created one. E.g.:
[image]

edriouk, 2018-03-01:
I assume you are using a local pack that is not published (yet) at www.keil.com/pack (see http://www.keil.com/pack/doc/CMSIS/Pack/html/createPackPublish.html#cp_KeilComPack). Publishing process creates then the device URL as well. That allows users of development tools (not only from ARM)  see and install your pack.

For the time being the code that displays the URL does not know if the pack has been published or not. 
We could considre fixing it in the next release (not scheduled yet)