ARM-software/cmsis-pack-eclipse #63: RefClient updates for latest CMSIS pack Eclipse plugin?

radhika-raghavendran, 2018-05-17:
Hi,

I notice that there are frequent releases to the CMSIS pack Eclipse plugin, but they do not contain the RefClient plugin.
Is there a plan to update the RefClient plugin also in the next release?
Also, is the older version of the RefClient (v2.2.0) compatible with the latest CMSIS pack plugin (v2.3.2)? I would think the latest releases are not backward compatible, and wanted to confirm.

Thanks
Radhika

edriouk, 2018-05-17:
Hi Radhika,
RefClient is actually a "sandbox"  plug-in to demonstarate CMSIS plug-in features and the wasy to utilize them in a customized environment. Therefore RefClient is never included in the binary build. Its most value is in the source code, not binary and its backward compatibility is not considered to be an issues. 
The plug-in itself gets updatetd only to demonstrate new features or significant changes and thus only for those cases we publish its binary.

marguzmar, 2019-09-23:
Question was answered.

closed 2019-09-23 by marguzmar