ARM-software/cmsis-pack-eclipse #64: Example documentation

satishchandra-mindtree, 2018-06-11:
Hi All,

I am not sure how to get the document section of CMSIS Pack examples to get going. I added following line in the .pdsc file for my application example, but still I dont see the document.
<book category="manual"    name="documentation\application_reference.pdf" title="Application reference"/>

I have tried multiple other entries for documents, but still the document section of the project explorer of eclipse is empty.

Thanks,
Satish.

edriouk, 2018-06-12:
Hi Satish,

Since any example represents a project, you can add its documentation directly to its content as a regular project file. When the example is installed, the documentation will get extracted as well. 

The example element has an attribute "doc" that should point to a file relative to path specified in "folder" attribute.  The idea was to show a link to that documentation file in the "Examples" view of Pack Manager perspecive. However currently the "doc" attribute is not processed since it can be only evaluated when the pack is installed.

Best regards,
Evgueni

satishchandra-mindtree, 2018-06-12:
Hi Edriouk,

I was trying to modify the pdsc file of already installed pack. Will try creating new pack again and see.
Following is the example section of pdsc file,

    </example>
	    <example doc="README.md" folder="source/samples/rslx/BB_BLE2_Terminal" name="BB_BLE2_Terminal">
      <description>BB BLE terminal for RSL10</description>
      <board name="RSL10 Evaluation Board" vendor="ONSemiconductor"/>
      <project>
        <environment load=".project" name="rslx"/>
      </project>

      <attributes>
		<files>
			<file category="doc"    name="README.md"/>
		</files>
        <component Cclass="Device" Cgroup="Libraries" Csub="CMSIS Startup"/>
        <component Cclass="Device" Cgroup="Libraries" Csub="System"/>
        <component Cclass="Device" Cgroup="Libraries" Csub="Kernel"/>
        <component Cclass="Device" Cgroup="Libraries" Csub="BLE"/>
        <component Cclass="Device" Cgroup="Libraries" Csub="Weak_PRF"/>
        <component Cclass="Device" Cgroup="Bluetooth Profiles" Csub="BASS"/>
      </attributes>
    </example>

Thanks,
Satish.

edriouk, 2018-06-12:
Hi Satish,

Your example documenation README .md and/or README.doc should be part of the Eclipse project  in source/test directory in order to see it in the project explorer. 

Please note, that according to CMSIS documentation 
http://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_examples_pg.html#element_example_attributes
and PACK.xsd schema file thare is no "files" element in example attributes.

Best regards,
Evgueni

closed 2024-01-02 by jkrech