ARM-software/cmsis-pack-eclipse #65: Could not find file 'cmsis-pack-eclipse-master'

naveenkmedi, 2018-06-25:
I have a question(I don't know where to ask).
I am trying to install CMSIS pack on eclipse. I have downloaded the zip file and unzipped it. When I select the folder from eclipse Help>Install New software> it is showing an error "Could not find file 'path/cmsis-pack-eclipse-master'. I'm using windows 10.

Do I need to build or do something before installing on Eclipse?

JonatanAntoni, 2018-06-25:
Hi @naveenkmedi,

There should be no need to unzip the archive manually. The archive contains a software repository. You should be able to install the provided plugins with "Help > Install New Software... > Add... > Archive... > CmsisPackPlugIn2.3.2.zip". The list of "Available Software Sites" should state something like "jar:file:/CmsisPackPlugIn2.3.2.zip!/". If you select to "Work with" that source you should see "CMSIS Pack Eclipse plug-ins".

I tried to do this using fairly old Eclipse Mars CDT and it works. I am not sure if the plug-ins are already compatible with Neon, Oxygen or Photon.

Cheers,
Jonatan

naveenkmedi, 2018-06-25:
I've tried that also. It shows the similar error -  Could not find jar:file:/CmsisPackPlugIn2.3.2.zip!/.
I am using Neon.3 Release (4.6.3)

JonatanAntoni, 2018-06-25:
I can confirm that installing
- [Eclipse Neon.3 (4.6.3) with CDT on Windows 10x64](http://www.eclipse.org/downloads/download.php?file=/technology/epp/downloads/release/oxygen/3a/eclipse-cpp-oxygen-3a-win32-x86_64.zip) 
- [CMSIS Pack Plug-ins 2.3.2](https://github.com/ARM-software/cmsis-pack-eclipse/releases/download/v2.3.2/CmsisPackPlugIn2.3.2.zip)
works as expected. Some requirements have been fetched from official Eclipse update sites during install. CMSIS Pack Manager perspective works as expected.

May I ask you to try it again using above resources, please? Perhaps you have some sort of local clash with other plugins.

ReinhardKeil, 2019-02-22:
Closing as no further actives.  Issue could not be reproduced

closed 2019-02-22 by ReinhardKeil