ARM-software/cmsis-pack-eclipse #66: <vindex> section in .vidx files ignored

TTornblom, 2018-06-26:
I've been looking at setting up a .vidx file to list vendor pack repos, and I have been unable to get the <vindex> section to work.

I've used http://www.keil.com/pack/Keil.vidx to debug the issue, and the only packs that are listed in the plugin are the packs in the <pindex> section, the vendor sites listed in the <vindex> section appears to be ignored. I have verified that all listed vendors sites work.

Is this a bug/omission or have I misunderstood how it is supposed to work?

Looking at the specification at: https://www.keil.com/pack/doc/CMSIS/Pack/html/packIndexFile.html#vidxFile
I think it is pretty clear that the <pindex> section is only supposed to be used for vendors who have yet to migrate to the .pidx format used by the <vindex> section.

jkrech, 2018-06-26:
Hi Thomas, 
please take a look at the documentation here:
https://www.keil.com/pack/doc/CMSIS/Pack/html/createPackPublish.html
The way it is working is that the Pack Manager uses a "pre-compiled" flat list of latest pack description files (index.pidx).
It is the tool-vendors choice where this list is taken from and who compiles it.
The tools from Arm today use www.keil.com/pack/index.pidx
This list is automatically updated once a day the web server. The update process reads the file www.keil.com/pack/Keil.vidx which contains both <pdsc> and <pidx> tags.
In case of <pdsc> tags the attributes are used to download a copy of the pdsc. The file is imported in case of a new version and the index.pidx updated accordingly.
A <pidx> tag represents a vendor pidx file, containing a list of <pdsc> files hosted and maintained by this vendor. In this case the importer first downloads the latest version of the pidx file and then iterates through all <pdsc> tags.

Once all pdsc files have been updated, the importer generates a new index.pidx file including the latest versions of pdsc files. The Pack Manager downloads the  index.pidx and downloads those pdsc files which have a new version listed.

I do agree that the update functionality could be implemented within the Pack Manager, however having this done on the web server, avoids redundancy and unnecessary download of files.

Please let me know if you have further questions.

Regards,

Joachim

TTornblom, 2018-06-26:
Thanks Joachim, this confirms my suspicion.

We also have a process in place that once a day builds an index.idx file, although currently I manually update the file on our web, just to make sure that there are no surprises for non working packs.

Cheers,
Thomas

closed 2018-06-26 by TTornblom