ARM-software/cmsis-pack-eclipse #68: Support needed for new CMSIS syntax element "<o key-identifier>*"

RobertCrossman, 2018-08-01:
The CMSIS annotation has been expanded to include "\<o key-identifier\>* ".
Is there a plan to support this?

The annotation page: http://www.keil.com/pack/doc/CMSIS/Pack/html/configWizard.html#configWizard_annotations

edriouk, 2018-08-02:
Yes, the support for that comes to the next release (not scheduled yet).

RobertCrossman, 2018-09-10:
I have implemented support for the "<o identifier>" feature.  I've also forked the repository with the intent of contributing the code back thru a pull request.

Would you consider this?  If yes, do you have any requirements for contributed code?  I have commented my code, and will make several commits to my fork so I can explain/document my changes in smaller pieces.

edriouk, 2018-09-11:
Hi Robert,
Contributions are welcomed via pull requests. 
We do not have official requirements for the code apar that it should be properly commented, avoids  vendor-specific features and additional dependencies as well as does not confilct with existing functionality .

Have you implemented the support for <-o key-identifier> feature? If yes, it could conflict with our implementation that is already done, but not yet published.

edriouk, 2018-12-20:
Hi Robert,
The feature including your contribution is published in v2.4.0 release.

closed 2019-02-22 by ReinhardKeil