ARM-software/cmsis-pack-eclipse #73: fatal error: Board_LED.h: No such file or directory

m-abubakar, 2019-01-26:
I have created a simple project in eclipse. When I select either LED or Buttons from Board Support packages and compile my project I get an error such as :  
`fatal error: Board_LED.h: No such file or directory`
OR 
`fatal error: Board_Buttons.h: No such file or directory`

I am using CMSIS-Pack Eclipse Plug-in 2.4.0 with eclipse Version: 2018-12 (4.10.0)
Build id: 20181214-0600.

edriouk, 2019-01-27:
Please provide more information:
- used compiler
- Eclipse toolchain for that compiler (selected in the first wizard page for C/C++ project) and its version
- toolchain adapter used (selected on the third page of CMSIS RTE project C/C++, after device selection)
- target device 
Could you please also attach .cproject and *.rteconfig files from you project.

m-abubakar, 2019-01-27:
Compiler : ARM Cross GCC (gcc-arm-none-eabi-8-2018-q4-major)
toolchain adapter : GNU MCU Eclipse ARM Embedded GCC (arm-none-eabi-gcc)
target device is STM32F103RB

rteconfig: 
`<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<configuration xmlns:xs="http://www.w3.org/2001/XMLSchema-instance">
  <packages/>
  <device Dclock="72000000" Dcore="Cortex-M3" DcoreVersion="r1p1" Dendian="Little-endian" Dfamily="STM32F1 Series" Dfpu="NO_FPU" Dmpu="NO_MPU" Dname="STM32F103RB" DsubFamily="STM32F103" Dvendor="STMicroelectronics:13" info="ARM , 20 KB RAM, 128 KB ROM" url="http://www.keil.com/dd2/st/stm32f103rb">
    <package info="STMicroelectronics STM32F1 Series Device Support, Drivers and Examples" name="STM32F1xx_DFP" url="http://www.keil.com/pack/" vendor="Keil" version="2.3.0"/>
  </device>
  <toolchain Tcompiler="GCC" Toutput="exe"/>
  <components>
    <component Capiversion="1.0.0" Cbundle="NUCLEO-F103RB" Cbundleversion="1.7.0" Cclass="Board Support" Cgroup="LED" Cvendor="Keil" Cversion="1.0.0">
      <package name="STM32NUCLEO_BSP" url="http://www.keil.com/pack/" vendor="Keil" version="1.7.0"/>
      <file category="sourceC" name="Boards/ST/NUCLEO-F103RB/Common/LED_NUCLEO_F103RB.c"/>
    </component>
    <component Cclass="CMSIS" Cgroup="CORE" Cvendor="ARM" Cversion="5.1.2">
      <package name="CMSIS" url="http://www.keil.com/pack/" vendor="ARM" version="5.4.0"/>
      <file category="doc" name="CMSIS/Documentation/Core/html/index.html"/>
      <file category="include" name="CMSIS/Core/Include/"/>
    </component>
    <component Cclass="Device" Cgroup="GPIO" Cvendor="Keil" Cversion="1.3" deviceDependent="1">
      <package name="STM32F1xx_DFP" url="http://www.keil.com/pack/" vendor="Keil" version="2.3.0"/>
      <file category="header" deviceDependent="1" name="RTE_Driver/GPIO_STM32F10x.h"/>
      <file category="source" deviceDependent="1" name="RTE_Driver/GPIO_STM32F10x.c"/>
    </component>
    <component Cclass="Device" Cgroup="Startup" Cvendor="Keil" Cversion="1.0.0" deviceDependent="1">
      <package name="STM32F1xx_DFP" url="http://www.keil.com/pack/" vendor="Keil" version="2.3.0"/>
      <file category="include" deviceDependent="1" name="Device/Include/"/>
      <file attr="config" category="source" deviceDependent="1" name="Device/Source/system_stm32f10x.c" version="1.0.0"/>
      <file attr="config" category="header" deviceDependent="1" name="RTE_Driver/Config/RTE_Device.h" version="1.1.2"/>
    </component>
  </components>
  <apis/>
</configuration>
`
.cproject
`<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<?fileVersion 4.0.0?><cproject storage_type_id="org.eclipse.cdt.core.XmlProjectDescriptionStorage">
	<storageModule moduleId="org.eclipse.cdt.core.settings">
		<cconfiguration id="ilg.gnuarmeclipse.managedbuild.cross.config.elf.debug.1139924864">
			<storageModule buildSystemId="org.eclipse.cdt.managedbuilder.core.configurationDataProvider" id="ilg.gnuarmeclipse.managedbuild.cross.config.elf.debug.1139924864" moduleId="org.eclipse.cdt.core.settings" name="Debug">
				<externalSettings/>
				<extensions>
					<extension id="org.eclipse.cdt.core.ELF" point="org.eclipse.cdt.core.BinaryParser"/>
					<extension id="org.eclipse.cdt.core.GASErrorParser" point="org.eclipse.cdt.core.ErrorParser"/>
					<extension id="org.eclipse.cdt.core.GmakeErrorParser" point="org.eclipse.cdt.core.ErrorParser"/>
					<extension id="org.eclipse.cdt.core.GLDErrorParser" point="org.eclipse.cdt.core.ErrorParser"/>
					<extension id="org.eclipse.cdt.core.CWDLocator" point="org.eclipse.cdt.core.ErrorParser"/>
					<extension id="org.eclipse.cdt.core.GCCErrorParser" point="org.eclipse.cdt.core.ErrorParser"/>
				</extensions>
			</storageModule>
			<storageModule moduleId="cdtBuildSystem" version="4.0.0">
				<configuration artifactName="${ProjName}" buildArtefactType="org.eclipse.cdt.build.core.buildArtefactType.exe" buildProperties="org.eclipse.cdt.build.core.buildArtefactType=org.eclipse.cdt.build.core.buildArtefactType.exe,org.eclipse.cdt.build.core.buildType=org.eclipse.cdt.build.core.buildType.debug" cleanCommand="${cross_rm} -rf" description="" id="ilg.gnuarmeclipse.managedbuild.cross.config.elf.debug.1139924864" name="Debug" optionalBuildProperties="org.eclipse.cdt.docker.launcher.containerbuild.property.volumes=,org.eclipse.cdt.docker.launcher.containerbuild.property.selectedvolumes=" parent="ilg.gnuarmeclipse.managedbuild.cross.config.elf.debug">
					<folderInfo id="ilg.gnuarmeclipse.managedbuild.cross.config.elf.debug.1139924864." name="/" resourcePath="">
						<toolChain id="ilg.gnuarmeclipse.managedbuild.cross.toolchain.elf.debug.2109386142" name="ARM Cross GCC" superClass="ilg.gnuarmeclipse.managedbuild.cross.toolchain.elf.debug">
							<option id="ilg.gnuarmeclipse.managedbuild.cross.option.addtools.createflash.38026965" name="Create flash image" superClass="ilg.gnuarmeclipse.managedbuild.cross.option.addtools.createflash" value="true" valueType="boolean"/>
							<option id="ilg.gnuarmeclipse.managedbuild.cross.option.addtools.createlisting.210956070" name="Create extended listing" superClass="ilg.gnuarmeclipse.managedbuild.cross.option.addtools.createlisting"/>
							<option id="ilg.gnuarmeclipse.managedbuild.cross.option.addtools.printsize.107315340" name="Print size" superClass="ilg.gnuarmeclipse.managedbuild.cross.option.addtools.printsize" value="true" valueType="boolean"/>
[... cut ...]
		<device Dclock="72000000" Dcore="Cortex-M3" DcoreVersion="r1p1" Dendian="Little-endian" Dfamily="STM32F1 Series" Dfpu="NO_FPU" Dmpu="NO_MPU" Dname="STM32F103RB" DsubFamily="STM32F103" Dvendor="STMicroelectronics:13" info="ARM , 20 KB RAM, 128 KB ROM" url="http://www.keil.com/dd2/st/stm32f103rb"/>
		<files>
			<file name="RTE/Device/STM32F103RB/system_stm32f10x.c" version="1.0.0"/>
			<file name="RTE/STM32F101_102_103_105_107.dbgconf" version="1.0.0"/>
			<file name="RTE/Device/STM32F103RB/RTE_Device.h" version="1.1.2"/>
		</files>
	</storageModule>
	<storageModule moduleId="refreshScope"/>
	<storageModule moduleId="org.eclipse.cdt.make.core.buildtargets"/>
</cproject>
`

jkrech, 2019-01-28:
From the rteconfig file it looks like you have selected the component Cclass="Board Support" Cgroup="LED" with Capiversion="1.0.0", however there is nothing listed under <api> section. 
Technically the reason why the compiler does not find the Board_LED.h file is due to the fact that this file is located in the Keil.MDK-Middleware pack. Could you please trial installing that pack?

edriouk, 2019-01-28:
You need to install Keil.MDK-Middleware pack (version >=7.8.0), it contains the requied API files.

It is a bug in the CMSIS RTE dependeny solver: the missing API should have been reported during project save/update (though it could not report the originating pack for that API anyway).

Please also note that the Device Startup component does not containg a linker script for the device, you need to provide it yourself.

m-abubakar, 2019-01-28:
Thank you jkrech and edriouk, adding Keil.MDK-Middleware pack indeed solved the problem.

@ edriouk

> Please also note that the Device Startup component does not containg a linker script for the device, you need to provide it yourself.

Could you please point me towards some How-to or tutorial regarding that.
Can we use ldscripts generated by "gnu-mcu-eclipse" plugin project over here ?

edriouk, 2019-01-28:
Actually I do not have expertiese in writing linker script file. 
As a starting point you can take one from generic ARMCM3 device (create a project with that and select Device Starup component). 
The "gnu-mcu-eclipse" might also provide the initial script, but if I'm not mistaken, you need to use "gnu-mcu-eclipse"project templates first and then copy the script to a CMSIS RTE projects .  

Also googling for "linker script for STM32F1" brings some potensially useful links.

ilg-ul, 2019-01-28:
Yes, you can try to generate a F1 project with the GNU MCU Eclipse templates and take a look at the linker scripts.

However, please note that simply copying linker scripts from one project to another is not enough, the content of the linker script should be in sync with the startup code, since it uses symbols defined by the linker. 

And also the implementation of the `_sbr()` should match the inits in the startup and the linker script, otherwise dynamic allocation will fail. Dynamic allocation is used by newlib during inits if you use some of the standard I/O functions, so even if you do not explicitly call `malloc()`, `_sbrk()` may be called by newlib.

The conclusion is that the linker scripts, the startup code and `_sbrk()` must all be in sync for the application to start.

For C++ things are even more complex, since the startup code should also call the static constructors.

The autoritative information source for linker scripts is the GNU LD manual (in the `share/doc` folder of the toolchain).

If you have further questions, please use the GNU MCU Eclipse support forum.

m-abubakar, 2019-01-29:
Thank you ilg-ul, This is really helpful.

marguzmar, 2019-09-23:
Problem was solved and additional feedback was provided.

closed 2019-09-23 by marguzmar