ARM-software/cmsis-pack-eclipse #75: Run Eclipse headless with cmsis, throws exception?

mmedvedmtd, 2019-02-21:
Hi - Im actually trying to use Atollic TRUEStudio, 9.2.0, which has CMSIS pack installed.  When I try to run my headless build, I get a Java stack dump, even though my build runs just fine.  I know Eclipse can handle running in a system with no GUI (no X), but can CMSIS plugin?

Attaching the stack dump.
[eclipse_build_errors.txt](https://github.com/ARM-software/cmsis-pack-eclipse/files/2889217/eclipse_build_errors.txt)

edriouk, 2019-02-21:
Hi,

As far as I know the AtollicTRUEStudion 9.2.0 still uses version 2.0.1 of CMSIS plugins.  That version 
did not provide headless build support.

The CMSIS headless builder was first published in version 2.4.0 at the end of last year (see corresponding Release notes).  
The TRUEStudio needs to be upgraded to that version to use headless build.

ReinhardKeil, 2019-08-16:
Closing as head-less mode is now supported and no further activity on that issue.

closed 2019-08-16 by ReinhardKeil