ARM-software/cmsis-pack-eclipse #77: Pre_Include_Global incorrect path

mszczypior, 2019-04-16:
Hello, I am trying to use the Pre_Include_Global.h file, to do that I am adding correct component to .pdsc file:

<Pre_Include_Global_h> 
		#define XXX 1
</Pre_Include_Global_h>

I can see that Pre_Include_Global.h file is added to RTE directory, however the project cannot be build because of the incorrect path, it seems that plugin / eclipse is removing all slashes before the RTE:
cc1.exe: fatal error:  C:Usersmszczypiorworkspacexxx/RTE/Pre_Include_Global.h: No such file or directory

Can you please advise if there is any workaround / setting I could use to make it working?

edriouk, 2019-04-16:
Hi,
That is a strange behavior that I cannot reproduce. 
Do other command line options such as include paths with the same prefix "${workspace_loc:/${ProjName}}" , work correctly or have the same problem with missing backslashes? 
What toolchain do you use and which CDT version is used?

mszczypior, 2019-04-16:
Hi,
All other options are resolved correctly but they are placed in "" quotes. Example:
"${workspace_loc:/${ProjName}}/RTE" - works correctly
${workspace_loc:/${ProjName}}/RTE/Pre_Include_Global.h - problem occurs.

When I manually added the quotes to the Pre_Include_Global.h then it works fine.

[image]

Toolchain: GNU ARM Cross and Gnu Make Builder 

[image]

Thanks!

edriouk, 2019-04-16:
Hi,
I pushed the fix that adds quotes to the "Include files" option. 
I also created a new release 2.4.1 that contains this fix and some other bug fixes.
Please try.

Evgueni

mszczypior, 2019-04-17:
Hi, it works, thanks!

One more thing, I noticed that if you are using multiple components, each with Pre_Include_Global_h, and if in any of them, there are multiple defines, additional spaces are added in the Pre_Include_Global.h file. 
Example:

PDSC file:
			<Pre_Include_Global_h> 
				#define AAAA				1
			</Pre_Include_Global_h> 

			<Pre_Include_Global_h> 
				#define CCCC				1
				#define DDDD				1
            	#define EEEE				1
				#define FFFF				1
			</Pre_Include_Global_h> 
			
			<Pre_Include_Global_h> 
				#define BBBB				1
			</Pre_Include_Global_h> 

Pre_Include_Global.h file:
			
/* Single define in the component */
#define AAAA				1
/* Multiple defines in the component*/
#define CCCC				1
				#define DDDD				1
<-spaces??->	#define EEEE				1
				#define FFFF				1
/* Single define in the component */
#define BBBB				1

Could you please take a look at that?

mszczypior, 2019-04-17:
I apologise, github is cutting off the spaces. Here is the correct output (with comment instead of spaces)

/* Single define in the component /
#define AAAA 1
/ Multiple defines in the component*/
#define CCCC 1
<-spaces??-> #define DDDD 1
<-spaces??-> #define EEEE 1
<-spaces??-> #define FFFF 1
/* Single define in the component */
#define BBBB 1

edriouk, 2019-04-17:
Those spcases come from pdsc file. 
The pdsc parser takes the text as is, trims leading and trailing spaces and performs no further parsing or formatting.

That is pack creator responsibility to format the text in pdsc file (thought it coluld break xml formattting).

mszczypior, 2019-04-18:
Ok, thank you for the answer.

marguzmar, 2019-09-23:
Bug was fixed and second questions was answered.

closed 2019-09-23 by marguzmar