ARM-software/cmsis-pack-eclipse #79: RTE_Device.h - dynamic generation of the file

mszczypior, 2019-04-26:
Hello,

I would like to ask if there is a possibility to dynamically generate the RTE_Device.h file depending on the used components. 
I.e. let's say that we have two components (two drivers) in the application and at some point we would like to add a third one. In that case the RTE_Device.h file would be updated so new configuration options would be added (to allow setting of the new third driver). So I do not need to always have all the configuration options inside the RTE_Device, but it would depend on the components which are selected to be used in the application.

Best Regards,
Michal

edriouk, 2019-04-26:
Hi Michael,

CMSIS provides two options to do that:

1. RTE_Components.h: this option has been available since the eraly plug-in versions
This file gets generated every time you add/remove/update components. 
The content of this file is constructed out of <RTE_Copmponents.h> elements of component description
By including RTE_Components.h in your RTE_Device. h you can have conditional compilation.
http://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_components_pg.html#RTE_Components_h

2. Pre-include files: that is a relatively new option available from version 2.4.0
Each component can specify a global (for the entire project) or/and local (for the component only) file to be included in every compilation.
http://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_components_pg.html#Pre_Include_Global_h
http://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_components_pg.html#Pre_Include_Local_h

I hope that helps.
Best regards,
Evgueni

mszczypior, 2019-04-26:
Hello Evgueni,

I am already using the conditional compilation. What I would like to achieve is to hide / remove the parts of the RTE_Device file related to the components which are not used in the applications. So these components are not visible in the configuration wizard. Example:

[image]

In the screen above you can see 4 components. And now, if the application is not using the USART component anymore, then I can remove it from rteconfig so RTE_Device.h is updated and this component is no longer visible in the CMSIS Configuration Wizard.

Is such thing possible?

edriouk, 2019-04-26:
Hello Michael,

We do not have such functionality. I believe it would be hard to implement:
 - generating RTE_Device.h might be tricky: it is a user-editable file and all user changes should be preserved.
 - the config wizard parser only processes meta-innformation in the specific comments, it does not use C/C++ frontend to make inclusions and exlusions. Enhancing that would also require significant effort.

mszczypior, 2019-04-26:
Hello, 

Thank you for the answer!

Michael

ReinhardKeil, 2019-08-16:
Closing as there is no further activity and questions are answered