ARM-software/cmsis-pack-eclipse #81: Enabling gpdsc support from STMCube failes to recognize the gpdsc file.

jlonnberg, 2019-05-10:
To reproduce- 
1.Start a new Cmsis project for STM32F746 and enable STMCubeFramework under device. The error log now shows that we've failed to load the gpdsc file. 
2. Generate the gpdsc from STMCube which should remove the error, but it still remains.

Debugging the issue shows that when ticking the STMCube tickbox, cmsis pack tries to locate and read the gpdsc file. As the file has not been generated yet, fGeneratedPacks in CpPackManager is registered as ["name of gpdsc", null]. Moreover, trying to add the gpdscwatcher to the file generates a NoSuchFileException thus failing to register a watcher to detect any changes in the gpdsc file. From this point on, any calls to loadGpdsc will return on

`if(fGeneratedPacks != null && fGeneratedPacks.containsKey(file))
 {
			return fGeneratedPacks.get(file);
		}`
returns null as the fGeneratedPack map contains the file id registered with a null thus preventing a new read operation of the gpdsc which combined with the failed gpdscwatcher prevents any recognision of updates and changes in the gpdsc file.

Everything work as intended when restarting Eclipse, as the maps and watchers are re-registered and the files now exist.

edriouk, 2019-05-10:
Thank you for finding this. Could you please correct the issue and supply a pull request?

jlonnberg, 2019-05-16:
I've found a work around for the issue by not registering the gpdsc files to fGeneratedPacks unless it actually exist. This will also solve the issue with the listener which then is activated when the .gpdsc actually exist on disc. However, saving the rteconfig before the gpdsc will leave the GUI red for the CubeOption until a new selection event occurs. For this I've found no solution.

edriouk, 2020-05-27:
Fixed in v2.6.0

closed 2024-01-02 by jkrech