ARM-software/cmsis-pack-eclipse #82: Unable to create new Cmsis C/C++ projects using the wizard

jlonnberg, 2019-05-10:
When creating a new cmsis project using the wizard, both Next> and Finish are greyed out in the wizard making it unable to create any new project from the gui. Issue occured after switching to the 2.4.1 plugins.
[image]

edriouk, 2019-05-10:
The wizard actually created an MBS project. I thought it is not your case. 
The strange thing here is selection of Arm CC toolchain adapter for IAR compiler.
Could you please debug what is going there.

singds, 2019-07-28:
I'm experimenting the same issue with a fresh installation on linux
Eclipse IDE for C/C++ Developers
Version: 2019-06 (4.12.0)
Build id: 20190614-1200
OS: Linux, v.4.15.0-54-generic, x86_64 / gtk 3.18.9
Java version: 1.8.0_212
Also downgrading to 2.4.0 does not solve.

edriouk, 2019-07-29:
Is IAR toolchain support installed as well? 
I believe the  problem is not in the CMSIS plug-in, but in IAR  usage of it. 
Could you please debug the problem and submit a pull request if you find that CMSIS plug-in needs some customization.

YakeDD, 2019-08-12:
I have the same problem.Have you found a solution?

ReinhardKeil, 2019-08-12:
What toolchain are you using?  Is this with the IAR EW-ARM integration?

YakeDD, 2019-08-12:
My project type is CMSIS C/C++ Project, and my Toolchains is only Gross GCC. 
Is there a problem with my setup?
Thank you

ReinhardKeil, 2019-08-12:
As you have not provided any information about your setup, it is hard to tell.
Also we don't know which step is not working.

I recommend that you open a new Issue, as this is about IAR integration.

ReinhardKeil, 2019-08-12:
See also https://github.com/ARM-software/cmsis-pack-eclipse/issues/86 as your problem might be similar

YakeDD, 2019-08-12:
I'm in a different situation than #86. 
I can select device, I can select adapter. But I can't click finish button after selecting the adapter.
I want to use eclipse with keil Uv5, where can I see the instructions?
Thanks!

ReinhardKeil, 2019-08-12:
Uv5 has RTE integrated. Why would you want to use Eclipse?

singds, 2019-08-12:
[image]
In my case, i don't have nothing about IAR installed. I can't press 'Finish' in all possible configurations.
I gave up tring to debug the extension code because i don't find a fast and straightforward way to setup a development environment (Java is not my main language).
Digging in the code i end up on RteTemplateCmsisProjectPage.java:238 line that states
    setPageComplete(complete); // TODO : implement
If i'm not wrong this function should enable the next arrow or the finish button. The TODO comment triggers me.
I don't fully get the program structure but could this be releated to the issue ?

ReinhardKeil, 2019-08-12:
Please describe your top-level problem.  You are talking here about some details. 

cmsis-pack-eclipse is not a stand-alone tool.  It allows other tool vendors to incorporate CMSIS-Pack management into third-party toolchains.

singds, 2019-08-12:
I think that my top-level problem is clear: jlonnberg described it in the first post. Also the issue's title explains it well: 'Unable to create new Cmsis C/C++ projects using the wizard'. I'm just adding some extra information abount my expirience with the same problem.
I described something about my setup in an old comment. 
I am available to provide specific information if needed.

ReinhardKeil, 2019-08-12:
What toolchain do you want to use?  You say UV5 which makes no sense to me.

singds, 2019-08-12:
I'm not YakeDD: i've not speack about Keil. I joined the conversation later.
The toolchain i'm using is the 'gcc-arm-none-eabi-8-2019-q3' but at this stage of the project creation i think eclipse is not aware about it. I expect to specify the toolchain path later.

nicdiamor, 2019-09-03:
Hi,
I'm experiencing the same problem: in the project creation the project type is "CMSIS C/C++ Project" instead of "CMSIS RTE C/C++ Project". 
[image]

If "CMSIS C/C++ Project" is selected, after choosing the target (STM32F411RE in my case) the Output states "elf" instead of "exe", and the project creation cannot continue.

[image]

Is this problem solved? My Eclipse installation is new and clean, only the cmsis-pack-eclipse plugin has been added (JRE and JDK are installed).

edriouk, 2019-09-04:
There is definitely a bug in the wizard that prevents to proceed with the selected toolchain adapter. 
Thank you for reporting it, the fix will be available in the next release. 

The bug is not related to the output type : elf is correct.
The real problem is that none of the toolchain adapters is fully suitable to the selected toolchain.
Please note that the Eclipse definition of toolchain is used:  a plug-in responsible to create command lines for compiler and linker and launch corresponding executables.
Two different Eclipse toolchains can use the very same compiler and linker executables.
In particular the following Eclipse toolchains use the same Arm GCC compiler:
- "Cross GCC" (cdt.managedbuild.toolchain.gnu.cross.exe.debug) - you try to use it,
    but it is not supported by CMSIS plug-ins, no adapter is available for it 
 - "ARM Crocc GCC" (ilg.gnuarmeclipse.managedbuild.cross.toolchain.elf.debug)
    supported by CMSIS plug-ins using the "Cross ARM GCC Toolchain Adapter "  

Therfore I would recommend you to install "ARM Cross GCC" toolchain plug-in, it is available here: 
https://github.com/gnu-mcu-eclipse/eclipse-plugins/releases/tag/v4.5.1-201901011632/

ilg-ul, 2019-09-04:
> I would recommend you to install "ARM Cross GCC" toolchain plug-in ...

The official install page is: https://gnu-mcu-eclipse.github.io/plugins/install/

edriouk, 2019-09-04:
Thank you Liviu for the right link.

mario-pi, 2019-09-04:
I have been working on a fix on this bug just yesterday. Apparently the problem was the RteTemplateCmsisProjectPage not being added as a listener to its own adapter selector widget.

I am not sure if I did the right thing, but I'll send a pull request asap so that you can evaluate my fix.

mario-pi, 2019-09-04:
Sorry about the multiple commits, I had to fix some annoying whitespace differences.
I have now submitted a pull request: https://github.com/ARM-software/cmsis-pack-eclipse/pull/87

marguzmar, 2019-09-23:
Bug was founded and fixed.

closed 2019-09-23 by marguzmar