ARM-software/cmsis-pack-eclipse #84: Add  book element to CMSIS pack dependent from a parent CMSIS-Pack without redefining the device

Rania-Chaibi, 2019-07-09:
Hi,

We created a CMSIS-Pack dependent on a parent CMSIS-Pack and should use the same device defined in the parent CMSIS-pack. How we can add specific documents related to the device in the new pack without redefining the device?
As I  understand  <book> tag should be under devices/family as shown in this link: https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_family_pg.html#element_book, but it seems that we need to redefine the device which is already defined in the parent CMSIS-Pack otherwise we get this warning *** WARNING M359: Family has no Device(s) or Subfamilies: 'RSL10 Series
Is there another way tho add books to a pdsc file?

Thanks,
Rania

jkrech, 2019-07-10:
Hi Rania,

there is no way to augment or extend the device description of a device being specified in packA in packB. In fact the device vendor is responsible to ensure that each device is only described within one pack.

May I ask you for the reason you would like to split content across "parent" and "child" pack?
I cannot immediately see any benefit but a lot of drawbacks. How do you make sure that parent and child used by programmers stays in sync?

What kind of books are we talking about? You could e.g. add a document to the device:startup component 
<file category="doc" ...  />

I hope this helps.

Joachim

Rania-Chaibi, 2019-07-10:
Hi Jkrech,

Thank you for the quick reply, the packB has the same device description, vendor as packA and depend on it, so we added it in requirements/packages, (and it also uses the device:startup component defined in PackA).
We want the books to show in the .rteconfig->Device tab->Device data books, and when PackB is installed it adds additional books (pdf files) to PackA, but we don't want to duplicate the device definition since it is the same.

Regards,
Rania

engsergiosoares, 2019-07-10:
Hi JKrech,

I to clarify Rania's question, on PackB, we add just want to add a few software components and examples.

The use case is:
We have PackA, which goes to all users by default and contains startup code, and standard software components.

For users that want additional (or exclusive) features, we provide PackB (which adds extra software components, a few extra new example projects for the same Evaluation Board defined in PackA and also enhanced variants of existing software components from PackA).

In order to not duplicate all the information from PackA in PackB, we make PackB dependent on PackA, using the <requirements/><packages/> tags and specify which version of the parent pack is compatible.

The point is, with the additional software components in PackB, we have to add extra documentation (.pdf documents) but we couldn't find a way to place them in the pack, so they show up in the GUI somehow. Do you have any suggestion?

edriouk, 2019-07-11:
Hi,
Each software component can have a child element <doc> where you can specify either a file (pdf, txt, etc.) or an URL (http:\\ , https:\\)   pointing to the documentation.  The link to file or URL is available in Description column against the component. Clicking on description will open the specified document.
You can also supply additional documentation with your examples

edriouk, 2019-07-11:
Additionally as Joachim mentioned you can add several documents to to a component as its files, e.g.
<file category="doc" name= docs/readme.txt attr="config">
<file category="doc" name= docs/ComponentDescription.pdf attr="config">

Please note attr="config". It causes the file to be copied to the project. That is currently required, because only local files are added to the project and displayed in the Project Explorer. In future we change the behavior to show all component files, not only local ones.

jkrech, 2019-07-11:
See https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_components_pg.html#element_file for documentation. Key is the category attribute https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_components_pg.html#FileCategoryEnum. Note that you can only have one doc file per component.

For books related to boards,  you can add these to the board description.
See: https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_boards_pg.html#element_board_book

For examples there is provision for documentation as well:
https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_examples_pg.html#element_examples

However this may not be used by any tools today. Instead these get added to the projects for the various toolchains.

I hope this helps.

closed 2024-01-02 by jkrech