ARM-software/cmsis-pack-eclipse #85: Need info about plugin dependencies

arch3d792, 2019-07-15:
Failed to install the CMSIS-Pack 2.4.1 plugin due to following missing repositories, as it seems:

> An error occurred while collecting items to be installed
session context was:(profile=epp.package.cpp, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,com.google.gson,2.8.2.v20180104-1110
No repository found containing: osgi.bundle,org.apache.commons.io,2.6.0.v20190123-2029
No repository found containing: osgi.bundle,org.eclipse.launchbar.core,2.3.0.201906102002
No repository found containing: osgi.bundle,org.eclipse.launchbar.ui,2.3.0.201906102002
No repository found containing: osgi.bundle,org.eclipse.tools.templates.core,1.1.0.201812111206
No repository found containing: osgi.bundle,org.eclipse.tools.templates.freemarker,1.0.0.201812111206
No repository found containing: osgi.bundle,org.eclipse.tools.templates.ui,1.1.1.201812111206
No repository found containing: osgi.bundle,com.sun.xml.bind,2.2.0.v201505121915
No repository found containing: osgi.bundle,org.eclipse.jdt.core,3.18.0.v20190522-0428
No repository found containing: osgi.bundle,org.eclipse.jdt.debug,3.11.100.v20180515-0929
No repository found containing: osgi.bundle,org.eclipse.jdt.launching,3.10.0.v20180606-1318

Or is my Eclipse version too old (Oxygen.3a Release 4.7.3a)? Would be nice to have some description (or maybe installation manual) providing an information about what Eclipse versions are supported or what 3rd-party plugins are required. Haven't found any.

edriouk, 2019-07-16:
I believe the problem is in your specific Eclipse installation. 
The plug-in is proven to work with different Eclipse versions starting from Mars and till the most recent 2019-6. Actually the readme.md says about version 2.4.0:
Platform: This release of the CMSIS-Pack Management for Eclipse has been tested on Eclipse 4.5.2 (Mars) and 4.6.3 (Neon).

We recommend to install CDT Eclipse variant.  You may also try to install C/C++ Development Tools to your Eclipse prior to installing the CMSIS plug-in

arch3d792, 2019-07-16:
> We recommend to install CDT Eclipse variant.

Actually I already have CDT variant installed. I have all main features of CDT 9.4.3.201802261533 from http://download.eclipse.org/tools/cdt/releases/9.4.

UPD
And now, after I switched off all update repositories in Eclipse except CDT and Oxygen, updated all its indexes from the repository (didn't change any installed components even!) and did restart Eclipse IDE, the CMSIS-Plugin installation started just fine. Seems like an Eclipse glitch.

Sorry for inconvenience, it works fine now. Thank you.

closed 2019-07-16 by arch3d792