ARM-software/cmsis-pack-eclipse #86: Cannot see CMSIS RTE after installing plugin

David-Regan, 2019-08-05:
I am working through the instructions on CMSIS packs. I am using Eclipse 2019-03. I downloaded CmsisPackPlugIn2.4.1.zip and installed it along with the refClient.
I then moved on to the step Creating RTE project however I do not see a project type of RTE.
This is what I see

[image]

jkrech, 2019-08-05:
CMSIS C/C++ project is fine. Enter a project name and advance to the next wizard step.

David-Regan, 2019-08-06:
Hi Joachim

I had already gone past that stage.
I get what seems to be a dead screen. It will not accept searches and the drop downs are blank

[image]

ReinhardKeil, 2019-08-06:
The information in this dialog is coming from CMSIS-Packs that should be installed on the computer. Actually you need DFP's (Device Family Packs) that describe devices.

If the dialog is empty in can have two reasons:
  - no DFPs are installed
  - the directory where the DFPs are located is not set

To specify the directory, open from the menu Window - Preferences - CMSIS Packs and specify the CMSIS-Pack root folder.

If you have no packs, I suggest you download some manually to get started from here: https://developer.arm.com/tools-and-software/embedded/cmsis/cmsis-packs

There is also a pack installer in the Eclipse framework, but again this requires some configuration.

ReinhardKeil, 2019-08-12:
Is your problem solved in the meantime?  
If so, can you share what was wrong so that we can improve...

David-Regan, 2019-08-12:
HI Reinhard

Sorry for not replying.
I did download the device family pack for STM32F4 and I was able to proceed further.

Thank you.

closed 2019-08-14 by marguzmar