ARM-software/cmsis-pack-eclipse #88: Overwriting of defined preprocessor symbols

Denker1208, 2019-09-25:
Hi at all,

this is my first time of writing an issue in GitHub. I hope this is the correct way of reporting a problem.
At the moment I use the Eclipse 2019-09 and have installt the newest CMSIS-Pack plugin (2.4.1) release.
My Problem ist, that I can't save defined symbols permanently. If I add the define for "CMSIS_BITPOSITIONS" for example in every configuration and language the preprocessor commands are changed correctly, but if I restart eclipse the symbols are resetted to "_RTE_" and "CORE_M4". Once it worked, that I could save the symbols , but only for one language at the active configuration.

These are my settings:
C/C++ Project
C++ Managed Build
CMSIS C/C++ Project with ARM Cross GCC
NXP 4317 Cortex-M4
Cross ARM GCC Toolchain Adapter
GNU MCU Eclipse ARM Embedded GCC (arm-none-eabi-gcc)

Sometimes the Finish button doesn't appear, but I think this problem is reported already.
If I create a none CMSIS-Project the defined symbols are stored correctly.

I hope you can help me. I have searched for answers, but didn't find any. Is there another position to change the symbols that I didn't find? Which update mechanism overwrites the defined symbols?

Best regards
Jens

edriouk, 2019-09-25:
Hi Jens,
You should put your defines _before_ **RTE** one. 
The plugin's toolchain adapter for Arm Cross GCC uses **RTE** as a marker and erases/replaces all entries  after it.  There was no other method to distinguish CMSIS defines from user ones. 
Currently it is possible to remove that limitation, but that improvement still waits for implementation (not scheduled yet).

Best regards,
Evgueni

Denker1208, 2019-09-26:
Hi Evgueni,

thank you for your help.
I can deal with tihis limitation and thanks for the explanation.

Best regards,
Jens

closed 2019-09-26 by Denker1208

midiwidi, 2023-01-31:
Is there any news about this problem. Is it still planned to fix this?
For me it was a pain for quite a while until I finally found this issue here. 
This hidden behavior is just unexpected and if you create a symbol it is automatically below the _RTE_ one. One has to actively move it up using 
[image]