ARM-software/cmsis-pack-eclipse #90: Improve the "Resolve Missing Packs" feature to trigger a web search of missing packs

engsergiosoares, 2019-10-23:
Hi folks,

I love the "Resolve Missing Packs" feature, as it makes it easier to our customers to find the external dependencies without having to manually find the packs.

Unfortunately, it seems this feature assumes the user/IDE has already updated the whole list of packs from the web (i.e. downloaded all .pdsc files), which is not always true and takes a lot of time to run. 

In our use case, we leave web updates disabled by default, so the "Resolve Missing Packs" button never works, unless the user clicks on "Check For Updates on Web" button beforehand (which takes a long time to download all the .pdsc files).

So, one suggestion to improve the usability, would be to make the "Resolve Missing Packs" feature trigger a web update of ONLY the missing packs that project needs.

For example, in some of our projects, we just need the ARM.CMSIS pack or the ARM.FreeRTOS. So, if the "Resolve Missing Packs" could automatically trigger a web search and download of ONLY these specific packs it would be much faster and make the experience smoother.

edriouk, 2020-05-27:
Fixed in v2.6.0

engsergiosoares, 2020-05-27:
Thank you very much, @edriouk . We'll give that a try!

closed 2020-05-27 by engsergiosoares