ARM-software/cmsis-pack-eclipse #93: Compatibility 2020-06

oskrypnikov, 2020-06-17:
Could not install plugin on new version of Eclipse

edriouk, 2020-06-18:
Could you please specify the Eclipse version and platform as well as the error message.

oskrypnikov, 2020-06-18:
[image]
[image]

edriouk, 2020-06-18:
That is not the problem specific to CMSIS plug-ins, but rather Eclipse or Eclipse CDT.
I have tried to install another software from Eclipse update site, the result is the same:
[image]

edriouk, 2020-06-18:
Update: the problem is in Eclipse for C/C++ 2020.06
The plug-ins are successfully installed in Eclipse for RCP and RAP

edriouk, 2020-06-18:
Here is the comment CDT team: 
The workaround is to do the installation while running Eclipse with Java >= 11. You can then revert back to using Java 8 once you are done installing.

edriouk, 2020-06-18:
The next version of CMSIS plug-ins will require JavaSE-11 as a minimum.

oskrypnikov, 2020-06-18:
Thank you very much. Installed JSE 14.0.1 and switched eclipse to this version by eclipse.ini file. Everything is OK.
added

-vm
C:/Program Files/Java/jdk-14.0.1/bin/javaw.exe

before "-vmargs"

closed 2020-06-18 by oskrypnikov