ARM-software/cmsis-pack-eclipse #94: Link project build configuration and .rteconfig components variants

engsergiosoares, 2020-06-26:
Hi guys,

There is one use case we would love to have on the CMSIS-Packs plugin (not if it's possible).
In our projects, we sometimes use different variants of the same component. For example, we may choose a low power variant of the Bluetooth Stack.

Before CMSIS-Packs, we would just create 2 build configurations: one would link to the the FULL library and the other one would link to the low power version of the library.

Now, with CMSIS-Packs, this is no longer possible, because the RTE Components (.rteconfig file) are not linked to build configurations.

Are there any plans to add support for this?

Thank you.

edriouk, 2020-06-29:
Hi Sergio,
we considered possibility to have separate/shared RTE configurations for each CDT build configuration from the very beginning.
We have decided then to use single configuration due to complexity reasons: 
 1. Eiting multiple configurations in a single editor or synchronizing several editors is difficult. How would you imagine such editor (with selection validation)?  
 2. Synchronization with CDT.  Accessing some information or setting values is not always possible via CDT public API. Some tricks are already used to do that.

Therefore we do not see possibility to support multiple configurations in the observable future.

engsergiosoares, 2020-07-08:
Thanks, @edriouk