ARM-software/cmsis-pack-eclipse #99: Schema version "1.7.2" not recognised, package Keil.STM32F0xx_DFP.pdsc ignored.

atilkowski, 2022-03-28:
Hi.
I have installed a 2.6.1 release of a plugin.
After running "refresh" the pdsc files are cached but afterwards many of them are ignored and not installed.
At least that is what happens with most of STM32 CMSIS files, it appears most Keil's pdsc files of STM32 were updated recently to new schema.

Schema version "1.7.2" not recognised, package Keil.STM32F0xx_DFP.pdsc ignored.
--
Alex

edriouk, 2022-03-28:
Hi, could you please provide more information, at least a screenshot of the error.
The plug-ins do not check schema by default processing for performance reasons. 
Therefore I wonder where this message come from

atilkowski, 2022-03-28:
No problem.
I attach a screenshot made after refresh and also txt of all "..ignored" messages.
If that could help, I use Windows 7, 32-bit, Eclipse release 2018-09, Java 1.8.0_281.

Interestingly, this whole thing worked for the last year (Keil.STM32F0xx_DFP.2.1.0.pdsc, uses schema version 1.6.0) till I decided to refresh yesterday. New Keil.STM32F0xx_DFP.2.1.1.pdsc file uses schema version 1.7.2.

Because of this incompatibility, whole processor family was removed so now I cannot even revert to old, working 2.1.0 version. 
If you need some further info - let me know.

[image]

[Ignored.txt](https://github.com/ARM-software/cmsis-pack-eclipse/files/8363391/Ignored.txt)

edriouk, 2022-03-28:
Hi,
that is not CMSIS-Eclipse plug-ins. 
Your screenshot shows "Embedded C/C++ CMSIS Pack (Experimental)" feature. That is absolutely independent developments that is completely separate from CMSIS Eclipse Plug-ins. 
Our CMSIS- Pack Manager perspective should look like as follows: 
[image]

closed 2022-03-28 by edriouk

atilkowski, 2022-03-29:
Thanks, that explains everything.

edriouk, 2022-03-29:
FYI, as far as I know the "Embedded C/C++ CMSIS Pack (Experimental)" feature is maintained by  https://mcuoneclipse.com

ilg-ul, 2022-04-10:
The plug-in is part of [Eclipse Embedded CDT](https://projects.eclipse.org/projects/iot.embed-cdt).

Support for 1.7.x was added some time ago, please update your Eclipse.